ARMmbed/mbed-os #2806: armcc compilation with mbed-os library

TaniaMirzin, 2016-09-25:
Note: This is just a template, so feel free to use/remove the unnecessary things

### Description
- Type: Bug
- Related issue:
- Priority: Blocker 

---------------------------------------------------------------
## Bug

**Target**
K64F

**Toolchain:**
ARM

Toolchain version: 5.06 update 3(ARMCompiler5.06u3)

**mbed-cli version:** 
(`0.8.8`)

**meed-os sha:**
(`bdab10dc0f90`)

Expected successful linkage when application compiled with mbed-os library using ARM compiler

Error message :
.build-mbed-os/mbed-os/hal/targets/cmsis/TARGET_Freescale/TARGET_MCU_K64F/TOOLCHAIN_ARM_STD/MK64FN1M0xxx12.sct", line 79 (column 8): Error: L6236E: No section matches selector - no section to be FIRST/LAST.

**Steps to reproduce**
1.Install ARMCompiler5.06u3 compiler.
2. Pull mbed-os from git.
3.Compile mbed-os as a library.
4.Create simple main application.(the location of the application is in same folder where mbed-os and .build-mbed-os)
5.Compile the application :mbed compile -t ARM -m K64F --source ../main/main.cpp  --source .build-mbed-os
----------------------------------------------------------------
@jenia81

yogpan01, 2016-09-26:
@TaniaMirzin @jenia81 You need an updated mbed-cli version, this one is quite old

TaniaMirzin, 2016-09-27:
@yogpan01 @jenia81 I used the latest mbed cli version and the problem wasn't solved.

TaniaMirzin, 2016-10-26:
The problem is still existing in 5.2 and with updated mbed cli.
Can you check it please

0xc0170, 2016-10-26:
@TaniaMirzin We will have a look. One more thing to clarify , this step : `` 3.Compile mbed-os as a library.`` what commands should we run?

I havent got any problem with armcc and mbed cli

adbridge, 2016-10-26:
@TaniaMirzin I just did the following:
```
  mkdir test_lib
  cd test_lib
  mkdir main
  (create main.cpp inside main directory)
  mbed-cli import mbed-os
  mbed compile -t ARM -m K64F --library --no-archive --source=mbed-os --build=../mbed-os-build
  mbed compile -t ARM -m K64F --source=main --source=../mbed-os-build --build=../basic-out
```
That compiled fine for me?

sg-, 2017-01-18:
@TaniaMirzin Can you confirm if the problem is resolved?

theotherjimmy, 2017-04-25:
@TaniaMirzin Bump. Is this resolved?

theotherjimmy, 2017-05-16:
No response. Closing. Please re-open if it's still relevant.

deepikabhavnani, 2017-09-21:
> No response. Closing. Please re-open if it's still relevant.
Closing :-)

closed 2017-09-21 by deepikabhavnani