ARMmbed/mbed-os #2820: Memory map summary doesn't work any more

jeromecoutant, 2016-09-27:
Hi

All values are kept to 0

Ex:
Memory map breakdown for built projects (values in Bytes):
+----------------------------+---------------+-----------+------------+-------+------+-----------+-------------+
| name | target | toolchain | static_ram | stack | heap | total_ram | total_flash |
+----------------------------+---------------+-----------+------------+-------+------+-----------+-------------+
| nist_internet_time_service | NUCLEO_F429ZI | ARM | 0 | 0 | 0 | 0 | 0 |
| tcp_client_echo | NUCLEO_F429ZI | ARM | 0 | 0 | 0 | 0 | 0 |
| tcp_client_hello_world | NUCLEO_F429ZI | ARM | 0 | 0 | 0 | 0 | 0 |
| udp_echo_client | NUCLEO_F429ZI | ARM | 0 | 0 | 0 | 0 | 0 |
+----------------------------+---------------+-----------+------------+-------+------+-----------+-------------+

adbridge, 2016-10-26:
@bridadan Do you know anything about this?

bridadan, 2016-10-26:
As far as I know this issue should be resolved. @jeromecoutant If you can verify this, could you please close the issue?

closed 2016-10-27 by 0xc0170

jeromecoutant, 2016-11-02:
Hi
It is at least OK with mbed-os-5.2.1
Thx