ARMmbed/mbed-os #2829: Provide mbed library support for STM discovery 407 board

sandeepach, 2016-09-27:
### Description
- Type: Bug 
- Priority: Major 
---------------------------------------------------------------
## Bug

**Target**
TARGET_DISCO_F407VG

**Toolchain:**
GCC_ARM|ARM|IAR - All, mainly interested in GCC_ARM and ARM

**Expected behavior**
Version 126 of mbed library should have the library for Target 407. However i dont see it. 
**Actual behavior**
Version 126 of mbed library should have the library for Target 407. However i dont see it. 
Only mbed source is having those directories. 

**Steps to reproduce**
Download the latest released version of mbed. It is seen that Discovery 407 is not there in the library.

ohagendorf, 2016-09-27:
Why do you thing that is not available? What did you do exactly? Or where did you look for?

You can find ever thing under 
```
hal\targets\hal\TARGET_STM\TARGET_STM32F4\TARGET_DISCO_F407VG
```
and
```
hal\targets\cmsis\TARGET_STM\TARGET_STM32F4\TARGET_STM32F407VG
```

I just tried a project export for the target and gcc_arm. It is possible to compile.

And the compatibility matrix (mbed export -S) looks OK.

sandeepach, 2016-09-27:
Hello Olaf
Yeah in mbed source is present but not in mbed version 126. 
@0xc170 - could you please help clarify why this is missing. Probably the targets.json

ohagendorf, 2016-09-27:
I used the current master branch. 
And with the current mbed-os-5.1 I've got the same result.

Please explain more in detail, what exactly is not working. And where.

sandeepach, 2016-09-27:
Hello Olaf
Can you please check developer.mbed.org/code. 
If you download the latest version of mbed library it doesn't have 407 discovery

sandeepach, 2016-09-27:
https://developer.mbed.org/users/mbed_official/code/mbed/ that is the full link. It has no entry for 407 discovery

ohagendorf, 2016-09-27:
That seems to be something concerning the online ide.

The DISCO_F407 was never part of the online ide and I'm sure will never be. This board (and also the never version DISC1_F407) is not mbed compatible: the new version doesn't have a virtual com port, the old version doesn't even have the MSD programming functionality. STM provides a description to hack the board to have the vcom function but it is necessary to solder two wires - possible but not easy. 

To use this target you have to use the offline toolchain. That is working without restrictions.

sandeepach, 2016-09-27:
Not just online ide. Why is it not a part of mbed library official version. This is something I can't understand.

sandeepach, 2016-09-27:
I am sure it is not concerning the online ide. I just need a gcc and arm version of the mbed library.

ohagendorf, 2016-09-27:
Then you can clone or download from https://github.com/ARMmbed/mbed-os and use it offline. That's the way we are using it. E.g. you can export a DISCO_F407 project with:
```
python project.py -m DISCO_F407VG -i gcc_arm -n MBED_BLINKY
```
and work with this. The result is copied to 
```
.build\export\MBED_BLINKY_gcc_arm_DISCO_F407VG
```

adbridge, 2016-10-26:
@bcostm @adustm Can you comment why DISCO_F407VG has no "releases" field in targets.json. Not being tagged as supported for mbed-2 is causing it to be omitted from the mbed compiled library. Thanks.

bcostm, 2016-10-27:
This platform SDK has not been done by ST but by the community. But if all mbed 2 tests are OK, I don't see any problem to add this releases field in the targets.json file.

adustm, 2017-01-03:
Hello,
@sandeepach do you still face this issue ?
@bcostm if this board is not mbed enabled as explained by @ohagendorf , can we still add the "releases" field in targets.json ?

@adbridge, as explained by bcostm, this board is not supported by st team. 
If someone has ever tested it and provides the test results here, we could think of adding "release" in targets.json.
Any volunteers ?

Cheers
Armelle

bridadan, 2017-01-03:
> @bcostm if this board is not mbed enabled as explained by @ohagendorf , can we still add the "releases" field in targets.json ?

That's a good question, I'm actually not sure of our official stance on that. Any thoughts @sg-?

sg-, 2017-01-18:
We don't have an easy way to validate MCU ports without a debug interface supporting CDC and MSC. https://www.mbed.com/en/about-mbed/mbed-enabled/

That is under consideration

closed 2017-01-18 by sg-