ARMmbed/mbed-os #2835: mbed-cli --usb not working

larsgk, 2016-09-28:
### Description
- Type: Bug
- Priority: Major

---------------------------------------------------------------
## Bug

**Target**
KL25Z

**Toolchain:**
GCC_ARM

**Toolchain version: arm-none-eabi-g++ (GNU Tools for ARM Embedded Processors) 4.9.3 20150529 (release) [ARM/embedded-4_9-branch revision 227977] **

**mbed-cli version: 0.9.5** 
(`mbed --version`)

**Expected behavior**
compiling with usb support

**Actual behavior**
```
[mbed] Auto-installing missing Python modules...
Building project test1 (KL25Z, GCC_ARM)
Scan: .
Scan: FEATURE_UVISOR
Scan: FEATURE_CLIENT
Scan: FEATURE_COMMON_PAL
Scan: FEATURE_BLE
Scan: FEATURE_IPV6
Scan: FEATURE_IPV4
Scan: FEATURE_STORAGE
Scan: mbed
Scan: env
Scan: usb
Compile: main.cpp
[ERROR] ./main.cpp:3:23: fatal error: USBSerial.h: No such file or directory
 #include "USBSerial.h"

```

**Steps to reproduce**
1. make a "hello world" main.cpp including "USBSerial.h"
2. build with `mbed compile -t GCC_ARM -m KL25Z --usb`

adbridge, 2016-10-25:
@larsgk Hi Lars, I will take a look and see if I can reproduce the same issue locally.

adbridge, 2016-10-25:
@larsgk 
Hi Lars , USBSerial is not supported in mbed-os-5 and by default I suspect this is what you are picking up. It is  supported in mbed-os-2 which you can easily try in the online IDE.
https://developer.mbed.org/handbook/USBSerial

Alternatively if you want to still use mbed-cli then the following steps should get you where you want. 
(Assuming your test program would be test_usbserial)
```
mkdir test_usbserial
cd test_usbserial
mbed config root .
mbed add https://developer.mbed.org/users/mbed_official/code/mbed/
mbed add https://developer.mbed.org/users/mbed_official/code/USBDevice/
mbed deploy

<copy/create your main.cpp into test_usbserial

mbed compile -t GCC_ARM -m KL25Z --usb
```

Please let me know if this fixes your issue.

larsgk, 2016-11-03:
thanks! - I'll try today :) - when will it land in mbed-os-5?

(also - I reported a series of USB bugs/enhancements ... can I ping someone to triage (or at least comment?) :)

adbridge, 2016-11-03:
@sg- Sam can you comment on when/if USB Serial will be supported in mbed-os 5 ?

@larsgk We're actively going through the issues on mbed-os currently so you should see some progress. Are there any particular issues which are blocking you ?

larsgk, 2016-11-03:
If i get usb working with your suggestion, i will try to do a crude WebUSB
implementation e.g.  inside usbcdc or usbdevice and do a PR or just attach
to the bug (would like to bring a working demo to chromedev next week).

One that is really annoying is the #2838 (block/halt everything when trying
to write and nothing connected...). In general: soft connect/disconnect and
notifications could be nice to get revisited by someone :)

Another one: if you made an easy API for adding custom headers, WebUSB
could (just) use that - #2839

sg-, 2016-11-03:
USB updates for mbed OS 5 are planned but not scheduled yet.

larsgk, 2016-11-04:
@sg-  ok - I hope I can hack a WebUSB solution with mbed (2 or 5) for some demos at chromedev over the weekend tho. Will try to get the mbed-os-2 working with my stuff & do some header fiddling over the weekend.  If I manage to get it in time, I'll put the changes in a PR or gist

closed 2017-01-18 by sg-

larsgk, 2017-01-18:
Just a post-close note:  it works nicely with mbed OS 2 + adding the library (of course).  Just posted this: https://github.com/empirikit/empirikit-controller

bridadan, 2017-01-18:
@larsgk Thanks for the follow-up post, glad you were able to make something awesome!