ARMmbed/mbed-os #2844: How to enable mbed TLS hardware entropy?

kjbracey-arm, 2016-09-28:
- Type: Bug / Question
- Priority: Major

**Target**
K64F, but a general issue

On master, mbed TLS HW entropy does not seem to be enabled any more for K64F. I can see something in a script that attempts to add an inclusion of platform_mbed.h to some config file to enable it based it on DEVICE_TRNG, but I don't see that in the default config.

Presumably it should be there?

Is the intent that all custom configs should include platform_mbed.h in the same manner as that script?

Or is that script supposed to be run during the build automatically?

kjbracey-arm, 2016-09-28:
@0xc0170, @mikaleppanen

teetak01, 2016-09-28:
I can verify that this has been broken. mbed Client applications are all failing registration with current head of MBED-OS master.

Defining this in our custom mbedtls config brings the functionality back:

+#include "mbedtls/platform/inc/platform_mbed.h"

@yogpan01 @TeroJaasko @marhil01

teetak01, 2016-09-28:
This is BLOCKER. @0xc0170

0xc0170, 2016-09-28:
Sorry for inconvenience, this was left yesterday evening, and this PR sent this morning should resolve it, please can you test: https://github.com/ARMmbed/mbed-os/pull/2843 . CI finished, it should be good to merge.

cc @andresag01

0xc0170, 2016-09-28:
The pull request referenced above was merged, this should be resolved now, please let us know if not

teetak01, 2016-09-28:
Thanks @0xc0170 , our tests are again Green vs. master. @kjbracey-arm can verify if their issue was also resolved.

closed 2017-01-19 by sg-