ARMmbed/mbed-os #2846: armcc default flags

TaniaMirzin, 2016-09-28:
Note: This is just a template, so feel free to use/remove the unnecessary things

### Description
- Type: Enhancement
- Related issue: armcc toolchain flags
- Priority: Minor

---------------------------------------------------------------
## Bug

**Target**
K64F

**Toolchain:**
ARM

**Toolchain version:**
5.06 update 3(ARMCompiler5.06u3)
**mbed-cli version:** 
0.9.5

**meed-os sha:**
 bdab10dc0f90 

----------------------------------------------------------------
## Enhancement
Code size is not optimized enough when compiled with mbed cli on arm cc toolchain. 
**Suggested enhancement**
Please consider to add "-Ospace" flag to DEFAULT_FLAGS in arm.py

TaniaMirzin, 2016-09-28:
@jenia81

sg-, 2017-01-19:
Looks like the small profile added this. https://github.com/ARMmbed/mbed-os/blob/4198222c54fa9c127f7e9252a7ae576ddedc5551/tools/profiles/small.json#L17

https://docs.mbed.com/docs/mbed-os-handbook/en/latest/advanced/toolchain_profiles/

closed 2017-01-19 by sg-