ARMmbed/mbed-os #2847: Interrupt handling on armcc toolchain.

TaniaMirzin, 2016-09-28:
Note: This is just a template, so feel free to use/remove the unnecessary things

### Description
- Type: Bug 
- Related issue: Interrupt handling on arm cc
- Priority: Major 

---------------------------------------------------------------
## Bug

**Target**
K64F

**Toolchain:**
ARM

**Toolchain version:**
5.06 update 3(ARMCompiler5.06u3
**mbed-cli version:** 
(`mbed --version`)
0.9.5
**meed-os sha:**
(`git log -n1 --oneline`)
 bdab10d  

**Expected behavior**
Proper handaling of  interrupts.
**Actual behavior**
When I try to work with button interrupt , the callback function is never activated.
**Steps to reproduce**
Please run the next code on K64F device:
volatile bool gIsStartIdentityFlow = false;static void identityButtonCb()
{
       printf("Button is pressed\n");
       gIsStartIdentityFlow = true;
       gIdentityFlowSuccess = false;
}
 
int main(int argc, char *argv[]){
 
InterruptIn identityButton(SW2);
identityButton.fall(&identityButtonCb);
 
}
When device is ready press SW2.
----------------------------------------------------------------

TaniaMirzin, 2016-09-28:
@jenia81

bridadan, 2016-09-28:
@TaniaMirzin Generally you shouldn't be using `printf` in interrupt context, as mentioned in the docs here: https://docs.mbed.com/docs/mbed-os-api-reference/en/5.1/APIs/io/InterruptIn.

Can you blink an LED from interrupt?

TaniaMirzin, 2016-10-05:
Thanks for your answer.But i tried not to use printf and it is still not working.

TaniaMirzin, 2016-10-05:
I tried it again and it is ok.Thanks.

closed 2016-10-05 by TaniaMirzin