ARMmbed/mbed-os #2848: How to properly add non initialized RAM section for ARM compiler?

nvlsianpu, 2016-09-28:
## Question

How to properly add non initialized RAM section for ARM compiler? I tried https://github.com/nvlsianpu/mbed/commit/173408cdffedb08b4a6862018b08e390ad9471a4 by creating additional section RW_IRAM0 but it was filled by 0xFFFFFF by a function sctterload_copy() which seams to be a ARM CC startup part.

nvlsianpu, 2016-09-28:
@pan- ?

0xc0170, 2016-09-29:
@AlessandroA You might know the answer?

AlessandroA, 2016-09-29:
Could it be related to [this](http://www.keil.com/support/docs/3480.htm)?

nvlsianpu, 2016-09-29:
No. The variable is size of 0xD8.

nvlsianpu, 2016-09-29:
Besides - it seams it is only side effect in my targets. I discovered that in fact some variable from .bss are initialized badly - etc. os_initilized from rt_CMSIS.c was initialized to 0xff.

nvlsianpu, 2016-09-29:
There is branch for fully reproduce my problem: https://github.com/nvlsianpu/mbed/tree/nrf5_noinit_bss_arm5_debug. In case an application is build ```mbed compile -m NRF52_DK -t ARM -o debug-info```
is visible that an app landed in hard fault - it is visible that app hadn't even get in application's main() and crash somewhere in kernel initialization.

sg-, 2017-01-19:
Looks like the link https://github.com/ARMmbed/mbed-os/issues/2848#issuecomment-250420999 and a forum post liked from that is confirmed to fix this. http://www.keil.com/forum/11937/

We also use this in DAPLink https://github.com/mbedmicro/DAPLink/blob/eab079b8a1326f0a3647e3df33cf5dda13994a1f/source/daplink/settings/settings.c#L54 and https://github.com/mbedmicro/DAPLink/blob/master/source/daplink/daplink.sct#L44
Could you try
```
+__attribute__ ((section("noinit"), zero_init))
```

closed 2017-01-19 by sg-