ARMmbed/mbed-os #2862: `mbed export` can't select the toolchain to generate a project for

Patater, 2016-09-29:
µVision supports projects using GCC or ARM. `mbed export` does not currently provide a way to generate a project that would use GCC (it is hard coded to generate a project that uses ARM).

To fix this issue, `mbed export` would be able to generate µVision 5 projects with the user's choice of GCC or ARM. Ideally, the existing `mbed compile -t` toolchain selecting option would be used consistently with both `mbed export` and `mbed compile`. For example, `mbed export -i uvision5 -t GCC_ARM -m K64F` would create a µVision  5 project that uses the GCC_ARM toolchain targeting K64F.

### Description
- Type: Enhancement
- Priority: Major

---------------------------------------------------------------
**mbed-cli version:** 
0.9.7

**mbed-os sha:**
b481da44e96763b39ba4b50709e410c04b381e4c

0xc0170, 2016-09-29:
cc @theotherjimmy @sarahmarshy

theotherjimmy, 2016-09-29:
@Patater Thanks for the feature request. We have been thinking about doing this for a bit. When you say µVision, do you mean 4 or 5?

PS. I like your Icon.

Patater, 2016-09-29:
@theotherjimmy I'm using µVision 5, but if µVision 4 can also work with either GCC or ARM, that'd also be a nice IDE to implement this feature for.

P.S. Thanks.

theotherjimmy, 2016-10-20:
@Patater We now have Make_xxxx and eclipse_xxxx exporters. We're getting there ;)

Patater, 2016-10-21:
@theotherjimmy Thanks for the update. Keep it up!

closed 2017-10-27 by ChiefBureaucraticOfficer