ARMmbed/mbed-os #2864: mbedtls breaks mesh thread for targets without Hardware Entropy Sources

hasnainvirk, 2016-09-29:
Note: This is just a template, so feel free to use/remove the unnecessary things

### Description
- Type: Bug 
- Priority: Blocker

---------------------------------------------------------------
## Bug

**Target**
OnSemi NCS36510

**Toolchain:**
GCC_ARM|ARM|IAR

**Description**

Newly commited changes to config.h in mbedtls now check if the platform have hardware entropy sources or not. If not, it includes a predefined file mbedtls/config-no-entropy.h". 
Mesh Thread use custom mbedtls config files for their proper operation and some of the supported platforms like OnSemi do not have Hardware Entropy Sources. However, still, there should be a hook to attach custom mbedtls user configuration file.

0xc0170, 2016-09-29:
@hasnainvirk - just to confirm, using latest master?

cc @sbutcher-arm @andresag01

hasnainvirk, 2016-09-29:
@0xc0170 Yes.

c1728p9, 2016-10-06:
This is the same as #2914 and is fixed by #2952

closed 2017-01-19 by sg-