ARMmbed/mbed-os #2883: LPC824: export to LPCXpresso, settings for release build gives compiler errors

JojoS62, 2016-10-02:
LPC824: export to LPCXpresso, settings for release build gives compiler errors

### Description
- Type: Bug
- Priority: Minor

---------------------------------------------------------------
## Bug

**Target**
LPC824, LPCXpresso824-MAX

**Toolchain:**
GCC_CR

**Toolchain version:**
LPCXpresse V8.2.0

**meed-os sha:**
729ef15

**Expected behavior**
**Actual behavior**
python project.py -m LPC824 -i lpcxpresso -n MBED_BLINKY
creates an archive that can be imported in the LPCXpresso IDE. 
Compiling in Debug mode works (when linker file is modified to use nano lib).
Switching to Release mode gives compiler errors because:
1) include path for C++ compiler is missing
2) linker settings include 'mbed' lib entry in libraries setting.

**Steps to reproduce**
- create export for LPCXpresso
- modify LPC824.ld to use nano lib:
GROUP(libgcc.a libc_nano.a libstdc++_nano.a libm.a libcr_newlib_nohost.a crti.o crtn.o crtbegin.o crtend.o)
- compile should report success
- switch to 'Release' configuration
- compile will report errors

to fix release build:
- copy all include path from C-compiler to C++ compiler
- remove 'mbed' from included libraries

----------------------------------------------------------------
## Enhancement
for using the nano lib, there is a linker option for GCC_CR in tools/toolchains/gcc.py, but this seems not to work as the linker file is fixed and not managed by LPCXpresso IDE.

theotherjimmy, 2016-10-03:
>for using the nano lib, there is a linker option for GCC_CR in tools/toolchains/gcc.py, but this seems not to work as the linker file is fixed and not managed by LPCXpresso IDE.

That's because the implementation of that particular section is incorrect. I'll solve the general problem, by allowing profiles for the exporters . This would allow you to pass `--profile mbed-os/tools/profiles/default.json --profile mbed-os/tools/profiles/nanolib.json` to the exporter.

theotherjimmy, 2016-10-03:
> I'll solve the general problem, by allowing profiles for the exporters . 

Past me already did this.

JojoS62, 2016-10-03:
That sounds great!

I’ve extended the exporters also for the LPC812 which is very similar to the LPC824.  Can I send you the necessary changes for this?

 

Thanks,

Johannes

theotherjimmy, 2016-10-03:
>I’ve extended the exporters also for the LPC812 which is very similar to the LPC824. Can I send you the necessary changes for this?

Yes! Do you have them on a branch on a fork? If you do, I can easily include you in the history.

JojoS62, 2016-10-03:
Yes, but I have only a local repo. A pull request is denied, so I do have to create one in my account? I’m not very familiar with working with git.

theotherjimmy, 2016-10-03:
Create a fork of this repo (there should be a button). Then add that fork as a remote to your local repo. Then push the branch to that remote.

JojoS62, 2016-10-03:
My copy was too old, there are a lot of changes now. I have cloned to a new repo and applied the changes. I can build the mbed lib for the LPC812 with GCC_CR, but I can’t export to lpcxpresso. 

 

D:\Projects\Sn\LocalGit\mbed-os\tools>python project.py -m LPC824 -i lpcxpresso MBED_BLINKY

Traceback (most recent call last):

  File "project.py", line 13, in <module>

    from tools.paths import EXPORT_DIR, MBED_BASE, MBED_LIBRARIES

ImportError: cannot import name MBED_BASE

 

What am I missing?

JojoS62, 2016-10-04:
I'm stepping into the exporters code. The previous problem was because I used the old repo which was not synced completely, so forget this.
The new fork works and have found already a reason for the double definition of the defines: there are two collections of flags, common_flags and c_flags/cpp_flags. The c/cpp_flags contain also the new defines in preprocesser (cpp) and symbols (c). I have removed them from the template and now the defines are only once in the settings.
Remaining: the -Os / -O0 flag is in this set, I would remove it for the Eclipse IDE because there is an own tab for 'optimization'. The 'other flags' override this optimization setting and as the result you change something in optimization but it is not working. I have commited the modification in my fork.
Now only the changed directory structure is messing up the exported pathes.

theotherjimmy, 2016-10-04:
Could you point me at the branch?

JojoS62, 2016-10-04:
https://github.com/JojoS62/mbed-os/blob/master/tools/export/codered_cproject_common.tmpl

theotherjimmy, 2016-10-04:
So, what you can do there is similar to what the IAR and uVision exporters do: You can check for the flag's presence and set a flag that is passed to the jinja template. If you do that, you should also remove the flag from the list of flags, again only when it's present.

This boils down to something like
```python
jinja_dict = {
...
    'optimization' : "-O2" in self.flags["c_flags"]
...
}
```
then later
```python
if  jinja_dict["optimization"]:
    jinja_dict["c_flags"].remove("-O2")
```
or similar.

JojoS62, 2016-10-04:
ok, I will check this tomprrow. Is the jinja or progen used for lpcxpresso?

theotherjimmy, 2016-10-04:
Jinja

theotherjimmy, 2016-10-04:
Thanks for looking into this @JojoS62!

JojoS62, 2016-10-04:
you're welcome, I like the mbed and it should be more popular! 
I have fixed now the inital problems, removed the mbed lib entry and added the include path to the c-compiler release build.

JojoS62, 2016-10-21:
re-opened because this problem still exists. 
I was able to fix the exporter and the modified codered_cproject_common.tmpl ist in my master branch of the forked project. My problem is now only how to use the exporter in general as mentioned in #2448 or #2995

theotherjimmy, 2016-10-21:
> I was able to fix the exporter and the modified codered_cproject_common.tmpl ist in my master branch of the forked project.

I'll take a look real quick.

theotherjimmy, 2016-10-21:
>I'll take a look real quick.

The changes look good.

sg-, 2017-01-19:
Has this been resolved?

JojoS62, 2017-01-20:
the support for LPCXpresso does not work at all now because it needs the GCC_CR which is deprecated now.
NXP has announced MCUXpresso, I think this will be the successor for  LPCXpresso and I hope it will be possibly to import LPCXpresso projects into MCUXpresso. 
So I suggest to close this issue and review it when MCUXpresso is availible. I see also the work on a new eclipse exporter, maybe this can be used for the LPC/MCUxpresso stuff also as they are eclipse based IDEs.

theotherjimmy, 2017-01-20:
Right. We could just change the LPCXpresso exporter to use GCC_ARM, which I think it does. If that's the case, we could get it up and running again real quick.

theotherjimmy, 2017-01-20:
> I see also the work on a new eclipse exporter, maybe this can be used for the LPC/MCUxpresso stuff also as they are eclipse based IDEs.

I hope so. It's shaping up nicely. So I'll close this issue now. @JojoS62 Feel free to open another issue, or re-open this one if something need updating or changing.

closed 2017-01-20 by theotherjimmy