ARMmbed/mbed-os #2884: LPC824: export to LPCXpresso, target running with wrong clock speed

JojoS62, 2016-10-02:
LPC824: export to LPCXpresso, target running with wrong clock speed

### Description
- Type: Bug

## Bug

**Target**
LPC824, LPCXpresso824-MAX

**Toolchain:**
GCC_CR

**Toolchain version:**
LPCXpresso V8.2.0

**meed-os sha:**
729ef15 Merge pull request #2682 from ohagendorf/bugfix_can_loop_test

**Expected behavior**
System clock should be set to 30 MHz.

**Actual behavior**
System clock is 12 MHz.
**Steps to reproduce**
export blinky to LPCXpresso, use Blinky test program. Blink frequency is wrong by factor 2.5 (30 / 12).

## Enhancement

**Suggested enhancement**
SystemInit() is not called in startup_LPC812_CR.cpp because of missing define:
```c
#if defined (__USE_CMSIS) || defined (__USE_LPCOPEN)
// Declaration of external SystemInit function
extern void SystemInit(void);
#endif
```
because of missing _USE_CMSIS define.
Solution:
_USE_CMSIS could be added in the project defines. But this define is used only once here. Maybe another existing define could be used for this purpose?

sg-, 2017-01-19:
Do you mind making a pull request with this fix?

JojoS62, 2017-01-20:
I have commited the fix to my forked repo. (How) can I send the PR for this single commit? There are other committs that are not necessary or confusing?

theotherjimmy, 2017-01-20:
Assuming that `armmbed` is the name of the remote corresponding to this repo, and that `origin` point to your fork:
1) `git checkout armmbed/master -b fix-lpcxpresso-clock` 
2) `git cherry-pick 77fc7dc1c`
3) `git push origin`
4) go to https://github.com/ARMmbed/mbed-os
5) Push the compare and PR button for `JojoS62:fix-lpcxpresso-clock` branch

I hope that helps.

JojoS62, 2017-01-20:
ok, that worked, hope I've done it right because this is my first PR.

theotherjimmy, 2017-01-20:
Yeah! It looks good.

JojoS62, 2017-07-24:
This issue is fixed and meanwhile GCC_CR has become obsolete. The startup code in GCC_ARM toolchain includes a call to SystemInit and this toolchain should be used.

closed 2017-07-24 by JojoS62