ARMmbed/mbed-os #2885: STM32L4: I2C driver not supporting custom CPU frequencies

TheTheAG, 2016-10-03:
The I2C driver for STM32L4 seems not to support custom CPU frequencies. Method i2c_frequency in file hal/targets/hal/TARGET_STM/TARGET_STM32L4/i2c_api.c expects SystemCoreClock to be either 80 MHz or 48 MHz. If it's something else, the I2C SCL frequency won't be changed at all.

I originally run into this using an older version of the driver which assumes the I2C clock to be 32 MHz but won't refuse to work with other frequencies. Since everything seemed to be configured automatically (from application programmer's point of view), I assumed that if I lower the CPU clock frequency from the default 80 MHz in order to save power, the I2C SCL frequency would remain more or less unchanged. It didn't.

I guess this is more an undocumented feature than a bug, but it still became an issue, and so I am reporting it.

0xc0170, 2016-10-03:
cc @bcostm @adustm @LMESTM

LMESTM, 2016-10-03:
hi @TheTheAG  - thanks for reporting this point. 
I need to update I2C driver for supporting I2C ASYNCH feature, so I'll have a look at the frequency management at the same time

LMESTM, 2017-01-26:
@TheTheAG 
on L4 devices, I had to make a choice on the possible supported clock frequencies.
This is implemented here
https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_STM/TARGET_STM32L4/i2c_device.h#L52
The I2C settings can be adapted very precisely on this family and specific setting can be computed thanks to STM32CubeMX program. I have now added support for systemCoreClock @ 80MHz and @ 48MHz, this should be ok for your need. If so let us know so that we can close the ticker

LMESTM, 2017-01-26:
@sg - I guess this can be closed unless @TheTheAG tells us otherwise

bridadan, 2017-01-27:
Thanks @LMESTM!

@TheTheAG Please reopen this if you'd like to revisit the issue!

closed 2017-01-27 by bridadan