ARMmbed/mbed-os #2890: Question: IAR with optimisation

RobMeades, 2016-10-03:
I have my new target working under GCC_ARM and have now added ARM and IAR toolchain support.  ARM works but I'm having trouble with IAR when optimisation is on.

The target runs fine with `-Ol`, the problem occurs when I move to `-Om`.  When I load code built with `-Om` the target is so broken that GDB can't even reset it (`WARNING: Failed to reset CPU. VECTRESET has confused core`).  The target only works if I compile *all* of the following directories with less than `-Om` optimisation:

```
mbed-os\hal\targets\cmsis
mbed-os\hal\targets\hal
mbed-os\hal\rtos
```

Just one of these directories makes no difference, it has to be all three, which makes me think the problem must be in a header file.  However, the compiler won't let me introduce `#pragma optimize=` lines in a header file so I can't narrow it down that way.

The IAR manual says that `-Om` introduces:

- Live-dead analysis and optimization
- Code hoisting
- Register content analysis and optimization
- Common subexpression elimination

Only "common subexpression elimination" can be suppressed individually and suppressing that makes no difference.

My question is: are there any "usual suspect" things I should look out for with IAR optimisation or are there any other suggested debug techniques before I start flashing an LED and binary chopping through the code?

Rob

RobMeades, 2016-10-05:
I'll take that as a "no" then :-).

sg-, 2017-01-19:
Did you get to the bottom of this?

RobMeades, 2017-01-20:
I reckon I must have 'cos I don't appear to do anything special for IAR.  B*ggered if I can remember what fixed it now though.

closed 2017-01-20 by RobMeades