ARMmbed/mbed-os #2892: [Expoters] Export broken after reorganization

sarahmarshy, 2016-10-03:
Note: This is just a template, so feel free to use/remove the unnecessary things

### Description
- Type: Bug 
- Priority: Major 

---------------------------------------------------------------
## Bug

**meed-os sha:**
(`git log -n1 --oneline`)

**Actual behavior**
Import errors. Resolving those, name errors

**Steps to reproduce**
Use mbed export command

0xc0170, 2016-10-03:
The referenced sha did not break it but reorganization ? Because that SHA above is not related to the changed structure.

>  Import errors. Resolving those, name errors. Resolving that, linking errors at build time.

Can paste some error so others can reproduce and fix (or they might see different errors).

sarahmarshy, 2016-10-03:
I am just posting the SHA of my local mbed-os. Not the SHA of the culprit. Sorry, was in the middle of pasting them. THere are above now.

theotherjimmy, 2016-10-20:
We got this resolved.

closed 2016-10-20 by theotherjimmy