ARMmbed/mbed-os #2902: default_lib and ARMCC microlib - consistency

0xc0170, 2016-10-04:
### Description
- Type: Enhancement 
- Reference: https://github.com/ARMmbed/mbed-os/pull/2210 (partially for the very old code-base, describes the problem).
- Priority: Major 

---------------------------------------------------------------
## Bug

**Expected behavior**

As GCC_ARM is one toolchain and provides 2 default_libs to be selected, ARMCC does not. You either have to select uARM or ARM. We might want to consider to unify this and have ARM only, and a target defines what default toolchain to support. This is also dragged into exporters where you have uvision (which one to use? ARM or uARM).

One related note for this unification, most of our targets (I guess all) provides separate linker script + startup file for microlib. I would assume we could unify  this as well, and any ARM supported target could be easily added to microlib.

**Actual behavior**

ARM or uARM to be selected as toolchain.

@theotherjimmy @sarahmarshy

0xc0170, 2018-06-14:
I am closing this as no activity for years.

closed 2018-06-14 by 0xc0170

theotherjimmy, 2018-06-14:
@0xc0170 #7197 resolves this issue.