ARMmbed/mbed-os #2903: Error when compiling mbed lib for LPC1768

wvdvegt, 2016-10-04:
Note: This is just a template, so feel free to use/remove the unnecessary things

### Description
- Type: Bug 
- Priority: Blocker

Get the following error after upgrading the mbed lib from build 26 to 127 yesterday.

Any suggestions? Error: "mbed::Stream::Stream(const mbed::Stream &)" (declared at extras/mbed_25aea2a3f4e3/Stream.h', '66'); return false;">/extras/mbed_25aea2a3f4e3/Stream.h:66</a>) is inaccessible in "extras/mbed_25aea2a3f4e3/Serial.h", Line: 50, Col: 43

Just defining an USB serial port seems enough to see the problem (the mbed lib and thus the program does not compile).

---------------------------------------------------------------
## Bug

**Target**
LPC1768

**Toolchain:**
Online

**Toolchain version:**

**mbed-cli version:** 
(`mbed --version`)

**meed-os sha:**
(`git log -n1 --oneline`)

**DAPLink version:**

**Expected behavior**
Compilation

**Actual behavior**
Error

**Steps to reproduce**
Just a small program using the serial port with LPC1768 as target should do.

wvdvegt, 2016-10-04:
Some more info (tried to compile with LPCXExpresso and got more errors

\Interpreter_copy\mbed/Stream.h: In copy constructor 'mbed::Serial::Serial(const mbed::Serial&)':
\Interpreter_copy\mbed/Stream.h:66:5: error: 'mbed::Stream::Stream(const mbed::Stream&)' is private
     Stream(const Stream&);

I'm trying to pass a Serial to my Cmdb command interpreter. It seems like the following constructor line is not compiling anymore:

Cmdb::Cmdb(const mbed::Serial& _serial, std::vector<cmd>& _cmds, void (*_callback)(Cmdb&,int)) :
        serial(_serial), cmds(_cmds) {

Switching to RawSerial is a workaround (code at least compiles again).

c1728p9, 2016-10-06:
It looks like your code is calling the copy constructor for the Serial class. This was made private a while back.

wvdvegt, 2016-10-07:
I do not get a warning on the serial copy constructor but an underlying class only (stream). 

If you make one private should the serial one not be private too? The online compiler output was not to helpful (off-line compiling made it more clear.

sg-, 2017-01-19:
@wvdvegt Can you publish a program with the problem. If will be faster to isolate and fix. If already resolved, please close the ticket.

closed 2017-10-27 by ChiefBureaucraticOfficer