ARMmbed/mbed-os #2907: make.py failing for option '-o'

mazimkhan, 2016-10-04:
uVisor debug builds are failing with following error (http://e108747.cambridge.arm.com:8080/job/mbed-os/job/mbed-os-pr-uvisor-test-pipeline/326/execution/node/60/log/):
```
12:19:24 [k64f_gcc_arm_d_0_0_0] Running batch script
12:19:24 
12:19:24 c:\jsbld\ws\mbed-os\mbed-os-pr-uvisor-test-pipeline@2\autogen\k64f_gcc_arm_d_0_0_0>sh build.cmd 
12:19:28 usage: make.py [-h] [-m MCU] [-t TOOLCHAIN] [--color] [--cflags CFLAGS]
12:19:28                [--asmflags ASMFLAGS] [--ldflags LDFLAGS] [-c]
12:19:28                [--profile PROFILE] [--app-config APP_CONFIG] [-p PROGRAM]
12:19:28                [-n PROGRAM] [-j JOBS] [-v] [--silent] [-D MACROS] [-S]
12:19:28                [-f GENERAL_FILTER_REGEX] [--automated] [--host HOST_TEST]
12:19:28                [--extra EXTRA] [--peripherals PERIPHERALS]
12:19:28                [--dep DEPENDENCIES] [--source SOURCE_DIR]
12:19:28                [--duration DURATION] [--build BUILD_DIR] [-N ARTIFACT_NAME]
12:19:28                [-d DISK] [-s SERIAL] [-b BAUD] [-L] [--rtos] [--rpc] [--eth]
12:19:28                [--usb_host] [--usb] [--dsp] [--fat] [--ublox] [--testlib]
12:19:28                [-l LINKER_SCRIPT]
12:19:28 make.py: error: unrecognized arguments: -o debug-info
12:19:28 [mbed] Working path "c:\jsbld\ws\mbed-os\mbed-os-pr-uvisor-test-pipeline@2\autogen\k64f_gcc_arm_d_0_0_0" (library)
12:19:28 [mbed] Exec "python -u c:\jsbld\ws\mbed-os\mbed-os-pr-uvisor-test-pipeline@2\autogen\k64f_gcc_arm_d_0_0_0\mbed-os\tools\make.py -t GCC_ARM -m K64F -o debug-info --source . --build .\.build\K64F\GCC_ARM -c -v" in c:\jsbld\ws\mbed-os\mbed-os-pr-uvisor-test-pipeline@2\autogen\k64f_gcc_arm_d_0_0_0
12:19:28 [mbed] ERROR: "python" returned error code 2.
12:19:28 [mbed] ERROR: Command "python -u c:\jsbld\ws\mbed-os\mbed-os-pr-uvisor-test-pipeline@2\autogen\k64f_gcc_arm_d_0_0_0\mbed-os\tools\make.py -t GCC_ARM -m K64F -o debug-info --source . --build .\.build\K64F\GCC_ARM -c -v" in "c:\jsbld\ws\mbed-os\mbed-os-pr-uvisor-test-pipeline@2\autogen\k64f_gcc_arm_d_0_0_0"
12:19:28 ---
```

Note: This is just a template, so feel free to use/remove the unnecessary things

### Description
- Type: Bug
- Related issue: 
- Priority: Major

---------------------------------------------------------------
## Bug

**Target**
K64F|??

**Toolchain:**
GCC_ARM|ARM|IAR

**Toolchain version:**

**mbed-cli version:** 
(`mbed --version`)

**meed-os sha:**
(`git log -n1 --oneline`)

**DAPLink version:**

**Expected behavior**

**Actual behavior**

**Steps to reproduce**

----------------------------------------------------------------
## Enhancement

**Reason to enhance or problem with existing solution**

**Suggested enhancement**

**Pros**

**Cons**

-----------------------------------------------------------------

## Question

**How to?**

theotherjimmy, 2016-10-04:
The `-o` options were replaced with `--profile`. In particular, `-o debug-info` became `--profile tools/profiles/debug.json`

mazimkhan, 2016-10-04:
@theotherjimmy will above option work with an ```mbed.json``` file in the app directory.

bridadan, 2016-10-04:
@mazimkhan As far as I know having that file in the app directory shouldn't cause any issues. The build profile system is separate from the config system, it won't automatically detect json files and use them. You have to use the `--profile` flag (though it does use `tools/profiles/default.json` if no `--profile` is specified).

mazimkhan, 2016-10-04:
@bridadan I meant will it still take config from the existing ```mbed_app.json``` file as that is required. If not how do we have both custom app config and debug flags?

bridadan, 2016-10-04:
@mazimkhan The systems are unrelated, so it should work just fine.

mazimkhan, 2016-10-04:
Ok. Thanks good then it will not require a big change to adapt. In future though we would like use just one way of supplying config. May abandon ```mbed_app.json``` and just use a profile json.

bridadan, 2016-10-04:
@mazimkhan Again, config and build profiles are totally separate things.

**Build profiles** are used to customize the compiler flags and optimizations (Note: **not macros**, which is what config does). This was always handled in the python scripts for the toolchains, @theotherjimmy's PR simply moves that config into modular files that you can edit and customize if you so desire. But for the most part, you should never need to do this. In fact, we probably discourage it as much as possible, but it is there now.

An effective use of the build profiles would be for enabling `gcov` flags for code coverage for instance.

**Config** is used for specifying macros, injecting extra labels, and enabling features so the build system pulls in different code. It affects the source tree walk and affects the `-D` flags that are passed to the compiler, but not the other compiler flags (`-g` for example can only be added by build profiles, **not** config)

mazimkhan, 2016-10-04:
Alright, got it now. Perhaps then **profiles** can have well known names that can be mentioned in the **Config** file just to keep build commands neat.

mazimkhan, 2016-10-04:
Closing as it was just a change in command.

closed 2016-10-04 by mazimkhan