ARMmbed/mbed-os #2908: STM32L053R8 export to Emblocks fails linking

janjongboom, 2016-10-04:
From: https://developer.mbed.org/questions/75168/Emblocks-export-doesnt-link/

> Hi, I would like to use emblocks from export mbed project ( STM32L053R8) but, when I try to generate .elf I have these log messages. It seems cannot link even though I export mbed-src. Can anyone help me? thanks
> 
> mbed-src\targets\cmsis\TARGET_STM\TARGET_STM32F4\stm32f4xx_hal_adc.c||In function 'HAL_ADC_MspInit':| mbed-src\targets\cmsis\TARGET_STM\TARGET_STM32F4\stm32f4xx_hal_adc.c|350|warning: unused parameter 'hadc' [-Wunused-parameter]| ticker.o: linker input file unused because linking not done| ..

theotherjimmy, 2016-12-20:
can we close this in favor of tracking the content on #2912  ?

janjongboom, 2016-12-21:
Sure.

closed 2016-12-21 by janjongboom