ARMmbed/mbed-os #2909: Cannot run anything on STM32L011: 'no space on device'

janjongboom, 2016-10-04:
Running 'Nucleo_blink_led' program (shows up when you create a new program for the L011). Compile it against any mbed library version (tried the last seven), compiles and links fine, but when flashing it fails.

janjongboom, 2016-10-04:
@screamerbg Can you ping ST?

0xc0170, 2016-10-06:
cc @bcostm @adustm @jeromecoutant @LMESTM

jeromecoutant, 2016-10-06:
Hi
As replied in https://developer.mbed.org/questions/75150/Impossible-to-download-mbed-bin-file-in-

I have tried Nucleo_blink_led with a NUCLEO_L011K4 with the inline compiler.
I didn't have any issue: compilation, flash, and blinking LED are OK

Note that only micro ARM is supported for this very small target.
.

hieuvt6-fti, 2016-11-28:
@jeromecoutant 
Could you please explain your process?

In my case, I do the following:
1. Launch the mbed Compiler (online)
2. Import Nucleo LED Blink example.
3. Hit compile
4. Wait for the file to download
5. Drag and drop the file into the Removable Device named NODE_L011K4

Then the copy will fail, complaining there is not enough space.

jeromecoutant, 2016-11-28:
Hi
It seems that issue us due to Windows version.
I am working with Windows 7, I think you are with Windows 10 ?
There is currently an issue with Windows 10 and very small target like L011K4

hieuvt6-fti, 2016-11-28:
@jeromecoutant
Indeed I am on Windows 10

bridadan, 2017-01-05:
Any update on this?

adustm, 2017-01-05:
Hi @bridadan 
The update of STLInk firmware for windows 10 should be available next week. Waiting for it

sg-, 2017-01-19:
@adustm Was the firmware released?

bcostm, 2017-01-20:
There is a new version V2.28.18 here:

https://my.st.com/content/my_st_com/en/products/embedded-software/development-tool-software/stsw-link007.license%3d1484918642757.html

Extract from the release note RN0093:
```
Add up to 20 Kbytes in virtual mass storage for disk size compared to the target flash
size, to have enough space after the host wrote some system files (for instance in a
Windows 10 environment)
```

So the issue should be solved with this new fw.

bridadan, 2017-01-20:
@janjongboom or @hieuvt6-fti are you able to confirm that the new firmware fixes the issue? I don't have a NUCLEO_L011K4 to test this myself.

janjongboom, 2017-01-23:
@andcor02 Could you check this? There's L011K4 under Mihail's desk.

andcor02, 2017-01-24:
@janjongboom @bridadan 

I can confirm upgrading the firmware has fixed the issue for Windows and Mac OS, able to drag binary to device, flash and run. 

Regards,

Andrea

closed 2017-01-24 by janjongboom