ARMmbed/mbed-os #2910: Wait() issue

Testato, 2016-10-04:
if you try to use the standard blink examples for mbed (NOT mbed-os)
and you use a 100ms of delay
You will found a bug in the wait function. the timing made many errors, the Led jump blinking

`
```
#include "mbed.h"

DigitalOut myled(PC_13);

int main() {
    while(1) {
        myled = 1;
        wait_ms(100);
        myled = 0;
        wait_ms(100);
    }
}`
```

0xc0170, 2016-10-06:
cc @bcostm @adustm  @jeromecoutant @LMESTM

bcostm, 2016-10-07:
Thanks for reporting this issue. I reproduced it on the Nucleo_F103RB.

I fixed the problem by adding these lines in the timer_irq_handler function in hal_tick.c file:
```
    if (__HAL_TIM_GET_FLAG(&TimMasterHandle, TIM_FLAG_CC3) == SET) {
        if (__HAL_TIM_GET_IT_SOURCE(&TimMasterHandle, TIM_IT_CC3) == SET) {
            __HAL_TIM_CLEAR_IT(&TimMasterHandle, TIM_IT_CC3);
        }
    }

    if (__HAL_TIM_GET_FLAG(&TimMasterHandle, TIM_FLAG_CC4) == SET) {
        if (__HAL_TIM_GET_IT_SOURCE(&TimMasterHandle, TIM_IT_CC4) == SET) {
            __HAL_TIM_CLEAR_IT(&TimMasterHandle, TIM_IT_CC4);
        }
    }
```

But I don't know why because channels 3 and 4 are not used... I continue my investigations. But in the mean time you can try it.

bcostm, 2016-10-12:
No, it is not related to CC3/CC4 interrupts.

It is related to the read of the timer master counter. I am going to send a fix soon.

Testato, 2016-11-22:
solved on mbed version 129
Thanks

closed 2016-11-22 by Testato