ARMmbed/mbed-os #2912: emblock

Testato, 2016-10-04:
Note: This is just a template, so feel free to use/remove the unnecessary things

### Description
- Type: Bug | Enhancement | Question
- Related issue: `#abc`
- Priority: Blocker | Major | Minor

---------------------------------------------------------------
## Bug

**Target**
K64F|??

**Toolchain:**
GCC_ARM|ARM|IAR

**Toolchain version:**

**mbed-cli version:** 
(`mbed --version`)

**meed-os sha:**
(`git log -n1 --oneline`)

**DAPLink version:**

**Expected behavior**

**Actual behavior**

**Steps to reproduce**

----------------------------------------------------------------
## Enhancement

**Reason to enhance or problem with existing solution**

**Suggested enhancement**

**Pros**

**Cons**

-----------------------------------------------------------------

## Question

**How to?**

theotherjimmy, 2016-10-04:
So, what's the issue exactly? this looks exactly like the template.

Testato, 2016-10-05:
Here some problematic Mbed export tests:
http://www.emblocks.com/forum/viewtopic.php?f=29&t=774

0xc0170, 2016-10-06:
I don't fully yet understand the points made in the forum discussions. We can discuss the problems here as they claim it's mbed related and it should be fixed within this repository. @Testato Can you summarize it here please?

cc @theotherjimmy @sarahmarshy

Testato, 2016-10-06:
Here i created two simply blink example, one for mbed-dev and one for mbed-os
If you try to export and import it on EmBitz (ex EmBlock) it is not possible compile it.
https://developer.mbed.org/users/testato/

theotherjimmy, 2016-12-20:
@Testato Would a project that calls out to make be okay with you? I can make that happen pretty quick for you.

Testato, 2016-12-21:
Sorry i do not understand your question, but in my repo you will found two examples that do not work.
You can use it for testing this issue
https://developer.mbed.org/users/testato/

theotherjimmy, 2017-01-03:
I was asking if it is possible for the EmBlock IDE to run `make`? Further, would generating a project file that runs `make` be acceptable?

theotherjimmy, 2017-01-03:
Actually, could you review #3521 @Testato ? I think it should fix you problem offline. Then I can work on getting the website up to date with it.

PR #3521 (2017-01-09): Repair the Emblocks exporer and rename to EmBitz
## Description
 * Renamed because the tool has been renamed
 * Removed duplicate symbol definitions
 * Removed common flags from ld flags (common are c/c++ only)
 * Moved EmBlitz to it's own folder

## Status
**READY**

## Migrations
Exporter name changed. No one needs to 'migrate' though.

## Related Issues
Resolves #2912 
Resolves #2736 

## Todos
- [x] Add Reviewers
- [x] Reviews
- [x] Manually check that a few exports build

## Steps to test or reproduce
`mbed export -i embitz -m ...`
closed 2017-01-09 by sg-