ARMmbed/mbed-os #2913: ST F746NG HAL is missing SAI enum after rev 121

janjongboom, 2016-10-04:
From https://developer.mbed.org/questions/74476/BSP-library-for-F746NG-not-working-with-/

The enum in mbed-dev\targets\cmsis\TARGET_STM\TARGET_STM32F7\stm32f7xx_hal_sai.h is missing a number of values since rev 121... The old listing is linked as an answer in the question.

0xc0170, 2016-10-06:
cc @bcostm @adustm @jeromecoutant @LMESTM

bcostm, 2016-11-08:
I don't know if the TIMEOUT and ERROR enums have been removed on purpose in the stm32f7xx_hal_sai.h but instead I have updated the BSP_DISCO_F746ZG library. The compilation on the mbed IDE is ok now.

We are going to update the CubeF7 HAL driver with a more recent version. We'll investigate this point with the SAI enums. If it was a mistake, I'll put back these enums in the BSP library.

Thanks for reporting that.

jeromecoutant, 2017-01-04:
Hi

STM32F7 drivers are OK,
and BSP_DISCO_F746ZG library is also OK thanks to Bruno's update,
so I think we can close this issue.

More information in mbed-os/targets/TARGET_STM/TARGET_STM32F7/device/Release_Notes_stm32f7xx_hal.html
 => V1.1.0 / 22-April-2016
    => HAL SAI update part

closed 2017-01-04 by 0xc0170