ARMmbed/mbed-os #2919: The global c++ class variable do not call constructor

meta-johnny, 2016-10-05:
Note: This is just a template, so feel free to use/remove the unnecessary things

### Description
- Type: Question
- Priority: Blocker | Major | Minor
---------------------------------------------------------------
## Question

I have a situation that declared a global c++ class which consist a constructor to initialize the variables inside the class.

But the code failed when I tried to access the variable which is initialized by constructor.

Any suggestion to make the global c++ class must call the constructor before the first time access the global class variable?

here is the example:

class Foo{

public: Foo():_data(1000){ // initialize _data to 1000
}

public: int getData(){ // return _data
    return _data;
}

protected:
  int _data;
};

class A : public Foo{ 
    public:
        A(): Foo(){
        }
};

A test;
int main(){
 // the data get is 0 not 100, any suggestion to make it right?
   printf("test->data = %d \n",test.getData());     
}

pan-, 2016-10-06:
@meta-johnny I've tested your example on the `NRF51_DK` with the following compilers: `ARMCC`, `GCC` and `IAR`.

What is your target ? Is it reproducible with your example or your issue is shown in a bigger application ?

meta-johnny, 2016-10-13:
@pan- From my example, I expected the print result is test->data = 100, but I got test->data =0.
It seem like the global class "A test" do not call the constructor to initialize the _data to 1000.

I just wonder when is the global class call the constructor? 
because it seem like it is not working in my project.

pan-, 2016-10-13:
@meta-johnny I will repeat myself but on what target and with which compiler this code fail ? 
I'm asking because I want to reproduce it. I'm not able to reproduce it on a target like the `NRF51_DK`.
It would also help if you can provide details about the mbed-os version (the git sha) that you use. 

> I just wonder when is the global class call the constructor? 

It is initialized before the main and it is compiler specific.

meta-johnny, 2016-10-14:
@pan- Thanks for your reply. I created my own target project based on mbed-5.1 and the compiler is armcc. 

Here is my compiler option:
arm-none-eabi-gcc.exe -mcpu=cortex-m4 -mthumb -Wextra -Wall -fdata-sections -ffunction-sections -fomit-frame-pointer -Os -g -funsigned-char -fno-exceptions -std=gnu99 
-Wno-unused-parameter -Wno-missing-field-initializers -fmessage-length=0 -fno-builtin -fno-delete-null-pointer-checks

Thanks.

pan-, 2016-10-14:
@meta-johnny I think there is few confusions here, you use GCC not the arm compiler. The compiler is not the target. The target is the board you compile for. Which board do you use ? 

And BTW, do you use mbed-cli to compile your project or another method ?

meta-johnny, 2016-10-17:
Sorry for my mistake.The compiler that I used is GCC. 
I tried my sample code on K64F board and it worked. Now I still clarify the root cause.
I do not used mbed-cli to compile my project but used EmBitz 0.42 instead. 
I have modified the compiler settings for my project development which based on mbed-OS 5.1.0-rc5. 
But I compare the settings with K64F it seem like the settings are almost the same.
Is this may caused by the environment settings? Do we need to initialize C++ environment?

pan-, 2016-10-17:
@meta-johnny Thanks for the clarification, that explains some things :).
With GCC, the initialization of global variables is made by calling the function `__libc_init_array` before the main. This is handled by the `C` runtime provided with your compiler or, if you compile with the RTOS, the override of the `pre_main` function ([here](https://github.com/ARMmbed/mbed-os/blob/5310c2e2cdfdbee49efa93ef51be5964c58a4549/rtos/rtx/TARGET_CORTEX_M/RTX_CM_lib.h#L591)). 

if you use mbed-cli, it will work just fine. Could you clarify the following points:
* What is the version of your compiler ? 
* Could you provide the command line of the link phase ? 
* Could you post the linker script you are using ? 

There must be a defect in one of those areas.

meta-johnny, 2016-10-18:
@pan Thanks for you help.
This is the version information of my compiler:
Using built-in specs.
COLLECT_GCC=arm-none-eabi-gcc
COLLECT_LTO_WRAPPER=c:/program\ files\ (x86)/gnu\ tools\ arm\ embedded/4.9\ 2015
q3/bin/../lib/gcc/arm-none-eabi/4.9.3/lto-wrapper.exe
Target: arm-none-eabi
Configured with: /home/build/work/GCC-4-9-build/src/gcc/configure --build=i686-l
inux-gnu --host=i686-w64-mingw32 --target=arm-none-eabi --prefix=/home/build/wor
k/GCC-4-9-build/install-mingw --libexecdir=/home/build/work/GCC-4-9-build/instal
l-mingw/lib --infodir=/home/build/work/GCC-4-9-build/install-mingw/share/doc/gcc
-arm-none-eabi/info --mandir=/home/build/work/GCC-4-9-build/install-mingw/share/
doc/gcc-arm-none-eabi/man --htmldir=/home/build/work/GCC-4-9-build/install-mingw
/share/doc/gcc-arm-none-eabi/html --pdfdir=/home/build/work/GCC-4-9-build/instal
l-mingw/share/doc/gcc-arm-none-eabi/pdf --enable-languages=c,c++ --disable-decim
al-float --disable-libffi --disable-libgomp --disable-libmudflap --disable-libqu
admath --disable-libssp --disable-libstdcxx-pch --disable-nls --disable-shared -
-disable-threads --disable-tls --with-gnu-as --with-gnu-ld --with-headers=yes --
with-newlib --with-python-dir=share/gcc-arm-none-eabi --with-sysroot=/home/build
/work/GCC-4-9-build/install-mingw/arm-none-eabi --with-libiconv-prefix=/home/bui
ld/work/GCC-4-9-build/build-mingw/host-libs/usr --with-gmp=/home/build/work/GCC-
4-9-build/build-mingw/host-libs/usr --with-mpfr=/home/build/work/GCC-4-9-build/b
uild-mingw/host-libs/usr --with-mpc=/home/build/work/GCC-4-9-build/build-mingw/h
ost-libs/usr --with-isl=/home/build/work/GCC-4-9-build/build-mingw/host-libs/usr
 --with-cloog=/home/build/work/GCC-4-9-build/build-mingw/host-libs/usr --with-li
belf=/home/build/work/GCC-4-9-build/build-mingw/host-libs/usr --with-host-libstd
cxx='-static-libgcc -Wl,-Bstatic,-lstdc++,-Bdynamic -lm' --with-pkgversion='GNU
Tools for ARM Embedded Processors' --with-multilib-list=armv6-m,armv7-m,armv7e-m
,cortex-m7,armv7-r
Thread model: single
gcc version 4.9.3 20150529 (release) [ARM/embedded-4_9-branch revision 227977] (
GNU Tools for ARM Embedded Processors)

Is this one the command line of the link phase? I deleted all the object paths. 
arm-none-eabi-gcc.exe -mcpu=cortex-m4 -Wl,-script="./hal/targets/cmsis/TARGET_META/TARGET_IVYXXX/TOOLCHAIN_GCC_ARM/IVYXXX.ld"  
-Wl,-Map=.\build\IVY_core_only.map   -Wl,--gc-sections -lstdc++ -Wall -Wextra -Wno-unused-parameter -Wno-missing-field-initializers -fmessage-length=0 -fno-exceptions -fno-builtin -ffunction-sections -fdata-sections -funsigned-char -MMD -fno-delete-null-pointer-checks -fomit-frame-pointer -Os -Wl,--wrap,main -Wl,--wrap,_malloc_r -Wl,--wrap,_free_r -Wl,--wrap,_realloc_r --specs=nano.specs   -L"C:\Program Files (x86)\GNU Tools ARM Embedded\4.9 2015q3\bin\..\arm-none-eabi\lib\armv7e-m" -L"C:\Program Files (x86)\GNU Tools ARM Embedded\4.9 2015q3\bin\..\lib\gcc\arm-none-eabi\4.9.3\armv7e-m\softfp" -L"C:\Program Files (x86)\GNU Tools ARM Embedded\4.9 2015q3\bin\..\arm-none-eabi\lib" -L"C:\Program Files (x86)\GNU Tools ARM Embedded\4.9 2015q3\bin\..\lib"   -lsupc++ -lm -lc -lgcc -lnosys 

This is my linker script.
[IVYXXX.zip](https://github.com/ARMmbed/mbed-os/files/535120/IVYXXX.zip)

Thanks

pan-, 2016-10-18:
@meta-johnny From what I see, the only difference is that `-lstdc++` is not included. Could you place a breakpoint at the entry of the function `__libc_init_array` ? It is the function which is supposed to call the constructors before the main. If this function is not available then, it has been removed from your executable. 

Please, also try to call it at the entry of the main and report if the data printed is the correct one.

In all case, the best thing you can do is to use the build system and linker scripts provided with mbed-os, otherwise you might expose yourself to unknown issues which certainly won't be fixed because we do **not** do this kind of support.

meta-johnny, 2016-10-25:
I checked my linker command, it is included -lstdc++. 
I tried to call __libc_init_array before main but Hardfault  error happened.
The libc.a we used is arm-none-eabi\lib\armv7e-m\libstdc++_nano.a.

Do you have any idea of what might cause this Hardfault issue happened?

meta-johnny, 2016-10-26:
Sorry. It is working now.
I called the __libc_init_array successfully by refer to K64F.
Thanks a lot.

pan-, 2016-10-26:
@meta-johnny Well, normally you don't have to call this function directly; the c runtime should do automatically it for you. There is still a pending issues in your setup. The best you can do is to use the tools supported by mbed, otherwise it will be impossible to support furthermore your case. 

PS: Please close this issue if you feel like it is resolved.

closed 2016-10-27 by meta-johnny