ARMmbed/mbed-os #2922: Latest mbed version 127 removed macros like DEVICE_SPI from source files

peeyushpashine, 2016-10-05:
What is the reason behind removing macros like DEVICE_SPI from source files? How can I know which macros will be defined for target in the makefile. Do I have a way to know target based macros apart from looking into the makefile? This is breaking peripheral support at my end.

peeyushpashine, 2016-10-12:
@bcostm Would you be able to clarify this?

bcostm, 2016-10-12:
I was not aware of such change. Can you give me more details where this macro has been removed ? I still see it in all spi_api.c files. In the targets.json file there is "SPI" written in the "device_has" field. I think this is converted in "DEVICE_SPI".

peeyushpashine, 2016-10-14:
I am sorry if I was not clear. from the device.h in board folders for the version 6c34061e7c34, DEVICE_SPI has value 1, but this statement is not present in the latest version device.h files for Nucleo. I see now in the makefile DEVICE_SPI macros are coming up. Is there a way for me to find out target board specific these macros the same way you are generating? If you are taking advantage of automation , I would like to do the same, otherwise it is tedious to go and add these macros in my makefile. @bcostm

bridadan, 2016-10-14:
The `DEVICE_*` macros specifically are generated from the target description here: https://github.com/ARMmbed/mbed-os/blob/master/targets/targets.json#L548

You can see the key `device_has` is an array of capabilities of the device. Any entry in that array will be translated to a compile-time macro with the name `DEVICE_<ENTRY>` where `<ENTRY>` is the value in the array.

closed 2017-01-04 by peeyushpashine