ARMmbed/mbed-os #2925: No export options for nRF52-DK yet

janjongboom, 2016-10-05:
Are we adding export options for nRF52 any time soon?

sarahmarshy, 2016-10-05:
#2708 should solve that by grabbing info from the CMSIS pack.

sarahmarshy, 2016-10-05:
(At least for Uvision)

theotherjimmy, 2016-12-20:
Looks like we got 'em.
```
+-----------------------+-------------+-------+-------+-------+----------+---------------+-----------------+-------------+----------+---------+-----+-----+------------+------------+--------------+----------+--------------+----------+---------+----------+-----+
| Platform              | atmelstudio | cmsis | coide | ds5_5 | e2studio | eclipse_armc5 | eclipse_gcc_arm | eclipse_iar | emblocks | gcc_arm | iar | kds | lpcxpresso | make_armc5 | make_gcc_arm | make_iar | simplicityv3 | sw4stm32 | uvision | uvision5 | zip |
+-----------------------+-------------+-------+-------+-------+----------+---------------+-----------------+-------------+----------+---------+-----+-----+------------+------------+--------------+----------+--------------+----------+---------+----------+-----+
| NRF52_DK              |      -      |   x   |   -   |   -   |    -     |       x       |        x        |      x      |    x     |    x    |  x  |  -  |     -      |     x      |      x       |    x     |      -       |    -     |    x    |    x     |  -  |
```

closed 2016-12-20 by theotherjimmy