ARMmbed/mbed-os #2927: Macros from mbed_app.json are not exported to IDE project

kuggenhoffen, 2016-10-05:
### Description
- Type: Bug
- Priority: Major

---------------------------------------------------------------
## Bug

**mbed-cli version:** 
0.9.7

**meed-os sha:**
2564a833c0200ca6aad5ca28b132cc8a80ce661d

**Expected behavior**
Macros defined in mbed_app.json are expected to get exported to IDE project files so that they are defined in build.

**Actual behavior**
Macros defined in mbed_app.json are not exported to IDE project files. I have tested with uvision5 and gcc_arm and neither result in working build if the macros are required.

**Steps to reproduce**
Add macro to mbed_app.json, export project, inspect project files and see that defined macro has not been exported.

nuket, 2016-10-10:
The macros defined in `mbed_app.json` do show up in `mbed_config.h` though, so as long as that file is included in the correct order the project will build.

Otherwise, I did have to add macros by hand to `C_FLAGS` and `CXX_FLAGS` in the `Makefile`.

0xc0170, 2016-10-10:
> The macros defined in mbed_app.json do show up in mbed_config.h though, so as long as that file is included in the correct order the project will build.

@kuggenhoffen Can you confirm ?

theotherjimmy, 2016-10-20:
@kuggenhoffen Hi Jaakko! Can we close this one? The macros should be there.

kuggenhoffen, 2016-10-20:
Hi all, seems to be working now as expected atleast for uvision5. I'll close this

closed 2016-10-20 by kuggenhoffen