ARMmbed/mbed-os #2931: SocketAddress.cpp ipv6_from_address does not clear byte who have been moved. FIX included

EduardPon, 2016-10-05:
In the function ipv6_from_address(), address values of IPv6 address after a :: is found are moved to the end of the short[] array. But the source memory of the moved part should also be cleared with zeros before the address values in front of the :: are copied. In the current implementation this may lead to an invalid address conversion. 
  
```
// Move suffix to end
memmove(&shorts[NSAPI_IPv6_BYTES/2-suffix], &shorts[0], suffix*sizeof(uint16_t));

// FIX, clear source of moved memory. 
for( int i = 0 ; i < NSAPI_IPv6_BYTES/2-suffix; i++ )
{
    shorts[i] = 0;
}
```

0xc0170, 2016-10-06:
cc @geky @kjbracey-arm

kjbracey-arm, 2016-10-06:
@mikaleppanen

mikaleppanen, 2016-10-06:
yes, there is a bug in that functionality

geky, 2016-10-06:
Thanks for reporting this! Must have been a pain to debug.

pr up with fix: #2953

I will add a set of unit tests for the address parsing functions.

PR #2953 (2016-10-25): nsapi - Fix leftover bytes from suffix during ipv6 parsing
Thanks to @EduardPon for hunting this down:

> In the function ipv6_from_address(), address values of IPv6 address after a :: is found are moved to the end of the short[] array. But the source memory of the moved part should also be cleared with zeros before the address values in front of the :: are copied. In the current implementation this may lead to an invalid address conversion. 

fixes #2931
closed 2016-10-25 by 0xc0170