ARMmbed/mbed-os #2933: K64F PinName for D8 is incorrect for REVD hardware

BlackstoneEngineering, 2016-10-05:
### Description
- Type: Bug 
- Priority: Major

---------------------------------------------------------------
## Bug
Pin D8 is incorrectly defined for REVD of K64F Hardware. 

**Target**
K64F

**meed-os sha:**
(`git log -n1 --oneline`)
24e1218

**Expected behavior**
According to the pinmap on 
[image] pin D8 maps to `PTC12`, but in the [PinNames.h](https://github.com/ARMmbed/mbed-os/blob/release/hal/targets/hal/TARGET_Freescale/TARGET_KSDK2_MCUS/TARGET_MCU_K64F/TARGET_FRDM/PinNames.h#L221) the pin is currently set to `PTA0`

BlackstoneEngineering, 2016-10-05:
@maclobdell @toyowata @sg-

BlackstoneEngineering, 2016-10-05:
Another win for the CI Test Shield!

0xc0170, 2016-10-10:
Resolved on master

closed 2016-10-10 by 0xc0170