ARMmbed/mbed-os #2936: K64F PwmOut - incorrect timing

BlackstoneEngineering, 2016-10-05:
Note: This is just a template, so feel free to use/remove the unnecessary things

### Description
- Type: Bug 
- Priority: Major

---------------------------------------------------------------
## Bug

**Target**
K64F

**meed-os sha:**
(`git log -n1 --oneline`)
release 5.1.4

## Problem
The PwmOut's on K64F are not behaving correctly. I used the following code
```c++
#include "mbed.h"

// Template to check Falling edge and Risign edge interrupts.
void main()
{
    PwmOut pwm(D3);
    pwm.period(0.5f);
    pwm.write(0.5f);
    
    while(1){;}
}
```
which you would expect to create a pwm wave with a period of 0.5s with a duty cycle of 50%. When measured the square wave had a frequency of 10.66ms. 
[image]

This behavior holds when I increase the period to 1s, instead it goes to 20.33ms.

## Solution?
Something is going wrong somewhere in the internal math. Haven't been able to find a solution yet.

BlackstoneEngineering, 2016-10-05:
@maclobdell

BlackstoneEngineering, 2016-10-05:
+1 for the CI Test Shield for finding this!

infinnovation, 2016-10-06:
IIRC the K64F's FTM uses a 16-bit counter with a 1MHz clock, so it doesn't handle a period longer than ~ 65ms.  Ideally the mbed code should do a range check and error out.  If you really need something that slow you could achieve it by other means.

BlackstoneEngineering, 2016-10-06:
Huh, thats weird. 

I can confirm this, I have tests running at 10ms and 30ms periods and they're passing, but the tests at 100ms are not. 

```
bedgt: test case summary: 9 passes, 4 failures
bedgt: all tests finished!
bedgt: shuffle seed: 0.1908070684
bedgt: test suite report:
--------------+---------------+---------------+--------+--------------------+-------------+
 target       | platform_name | test suite    | result | elapsed_time (sec) | copy_method |
--------------+---------------+---------------+--------+--------------------+-------------+
 K64F-GCC_ARM | K64F          | tests-api-pwm | FAIL   | 109.26             | shell       |
--------------+---------------+---------------+--------+--------------------+-------------+
bedgt: test suite results: 1 FAIL
bedgt: test case report:
--------------+---------------+---------------+-----------------------+--------+--------+--------+--------------------+
 target       | platform_name | test suite    | test case             | passed | failed | result | elapsed_time (sec) |
--------------+---------------+---------------+-----------------------+--------+--------+--------+--------------------+
 K64F-GCC_ARM | K64F          | tests-api-pwm | PWM_0 Frequency 100ms | 0      | 4      | FAIL   | 10.68              |
 K64F-GCC_ARM | K64F          | tests-api-pwm | PWM_0 Frequency 10ms  | 1      | 0      | OK     | 10.26              |
 K64F-GCC_ARM | K64F          | tests-api-pwm | PWM_0 Frequency 30ms  | 1      | 0      | OK     | 3.25               |
 K64F-GCC_ARM | K64F          | tests-api-pwm | PWM_1 Frequency 100ms | 0      | 4      | FAIL   | 10.69              |
 K64F-GCC_ARM | K64F          | tests-api-pwm | PWM_1 Frequency 10ms  | 1      | 0      | OK     | 10.26              |
 K64F-GCC_ARM | K64F          | tests-api-pwm | PWM_1 Frequency 30ms  | 1      | 0      | OK     | 3.26               |
 K64F-GCC_ARM | K64F          | tests-api-pwm | PWM_2 Frequency 100ms | 0      | 4      | FAIL   | 10.68              |
 K64F-GCC_ARM | K64F          | tests-api-pwm | PWM_2 Frequency 10ms  | 1      | 0      | OK     | 10.25              |
 K64F-GCC_ARM | K64F          | tests-api-pwm | PWM_2 Frequency 30ms  | 1      | 0      | OK     | 3.25               |
 K64F-GCC_ARM | K64F          | tests-api-pwm | PWM_3 Frequency 100ms | 0      | 4      | FAIL   | 10.69              |
 K64F-GCC_ARM | K64F          | tests-api-pwm | PWM_3 Frequency 10ms  | 1      | 0      | OK     | 10.26              |
 K64F-GCC_ARM | K64F          | tests-api-pwm | PWM_3 Frequency 30ms  | 1      | 0      | OK     | 3.26               |
 K64F-GCC_ARM | K64F          | tests-api-pwm | Pwm object definable  | 1      | 0      | OK     | 0.04               |
--------------+---------------+---------------+-----------------------+--------+--------+--------+--------------------+
bedgt: test case results: 4 FAIL / 9 OK
bedgt: completed in 109.51 sec
```

BlackstoneEngineering, 2016-10-06:
@sg- @0xc0170 Any thoughts about adding an upper and lower limit to the PWM API per board?

closed 2017-10-27 by ChiefBureaucraticOfficer