ARMmbed/mbed-os #2937: CPU Stop issue

Sim-jaeil, 2016-10-06:
SNote: This is just a template, so feel free to use/remove the unnecessary things

### Description
- Type: Bug | Enhancement | Question
- Related issue: `#abc`
- Priority: Blocker | Major | Minor

---------------------------------------------------------------
## Bug

```c++
#include "mbed.h"
Serial pc(PA_2, PA_3);

void Callback(){
 char Temp_Data = pc.getc();
 pc.putc(Temp_Data);
}

int main(){
 pc.attach(&Callback,Serial::RxIrq);
 while(1){
  
 }
}
```
I am currently using the device Nucleo_F401RE.
But even the same symptoms MBED 2.0, the CPU will stop the moment it takes an interrupt is received.
The compiler is okay to stop using the Microlib the Keil compiler is online and, Keil. It seems there is a problem with the MBED.

Until now Export by an external compiler, but resolved using Microlib The MicroLib there are additional libraries that do not work are looking for what to do now resolved.

Please help.

**Target**
NUCELO_F401RE

**Toolchain:**
ARM

**Toolchain version:**
5.06
**mbed-cli version:** 
0.9.7
**meed-os sha:**
(`git log -n1 --oneline`)

**DAPLink version:**

**Expected behavior**

**Actual behavior**

**Steps to reproduce**

----------------------------------------------------------------
## Enhancement

**Reason to enhance or problem with existing solution**

**Suggested enhancement**

**Pros**

**Cons**

-----------------------------------------------------------------

## Question

**How to?**

0xc0170, 2016-10-06:
cc @bcostm @adustm  @jeromecoutant @LMESTM

Sim-jaeil, 2016-10-07:
DISCO_F469NI board is also the same result.

bcostm, 2016-10-11:
For now I just confirm the issue on Nucleo_F401RE using your code above (BTW it was needed to rename the Callback function, otherwise the compilation failed).

Sim-jaeil, 2016-10-12:
Thank you for answer.

mbed IDE and the mbed-dev rev148: Do you use 21d94c4 was okay.
However, this behavior should not be using the latest version of mbed-os. It has not been updated yet?

In previous versions, replacing the callback function name to something does not work. Is there a way to solve something in a previous version?
Will you need to update your library?
gcc iar keil compilers include a lot of the characteristics for each another.

bcostm, 2016-10-12:
What I see GitHub release page:
mbed lib v127 = mbed-os-5.1.5 = [https://github.com/ARMmbed/mbed-os/commit/a6f3fd1a60d5df59246d7caf3f108c4d34e1808e]

and on the mbed IDE/revision page:
mbed-dev v148 = mbed lib v127 (written in the version comment, but I don't see explicitely the commit number corresponding to the v127...)

> In previous versions, replacing the callback function name to something does not work. Is there a way to solve something in a previous version?
Will you need to update your library?

I don't really understand what you mean but for me this is not related to the platform SDK. If I use the name "Callback" I have this message at compilation: 

`Error: "Callback" is ambiguous in "main.cpp", Line: 17, Col: 17`

I just renamed it to "myCallback" and it compiled...

Sim-jaeil, 2016-10-12:
thank you. The lack of English language skills, meaning does not go smoothly.
Using a Google Translator, I'm not sure this sentence is incorrect. ㅠㅜ

Below is a video of the test on the 5.1 version of mbed-os.
http://alconeng.synology.me:5000/fbsharing/CH3GcYY1
In previous versions, if a function name as myCallback this phenomenon.

Sim-jaeil, 2016-10-12:
Found out the cause of the phenomenon.
When including the RTOS library this will happen.
Please confirm.
MBED-Dev does not appear when only.

bcostm, 2016-10-13:
I just checked again with mbed lib v127 and everything is ok now to me... Can you please recheck also ?

Sim-jaeil, 2016-10-13:
[image]

I tried the check again.
But the result is the same. When you receive a letter from the serial port, CPU stops.
Can you check like this table?

Thank you

bcostm, 2016-10-13:
ok I used mbed library and not mbed-os library this is why.

Now I checked Nucleo_F401RE + mbed online ide + mbed-os (version 2541:a551faa = mbed-os-5.2.0-rc2) and program stopped when I receive a char on the serial port.

I don't know what happens when RTOS library is added...

I tried with mbed-os-5.1.5 and it is the same. I will check other older versions to see when the problem occured...

Sim-jaeil, 2016-10-21:
Do not have a solution yet?
[Serial] function -> [RawSerial] function ----- OK
Is two functions, what else?

bcostm, 2016-10-24:
No solution found.
I tried what you wrote with RawSerial and I confirm also that it works with this class.
But I have now a uVision project running and I can debug the program. For now what I see is a hard fault when a character is received. Don't know why yet. I continue my investigation...

bcostm, 2016-10-24:
The program crashs in function` _mutex_acquire` in file RTX_CM_lib.h. Looking in the assembly file, there is an hard fault when the assembly instruction `SVC `is executed.

Note that this function is called several times without any problem before a character is received.

Anybody has an idea ?

Sim-jaeil, 2016-10-25:
Does this problem occur for any other CPU not only for STM?

sg-, 2017-01-19:
The standard library that is part of ARM compiler 5 has mutex protection around stdio so you cannot use printf from interrupt context. https://developer.mbed.org/blog/entry/Simplify-your-code-with-mbed-events/ and https://docs.mbed.com/docs/mbed-os-api-reference/en/latest/APIs/tasks/rtos/

closed 2017-01-19 by sg-

Sim-jaeil, 2017-01-20:
I understand.
Thank you.