ARMmbed/mbed-os #2939: NUCLEO_F411RE - problems with InterruptIn object

BlackstoneEngineering, 2016-10-06:
### Description
- Type: Bug 
- Related issue: `#abc`
- Priority:  Minor

---------------------------------------------------------------
## Bug

**Target**
NUCLEO_F411RE, maybe more Nucleo boards?

**Toolchain:**
GCC_ARM

**Toolchain version:**
4.9.3 20141119

**mbed-cli version:** 
(`mbed --version`)
0.9.5

**meed-os sha:**
(`git log -n1 --oneline`)
24e1218

**Expected behavior**
CI Test Shield InterruptIn tests should pass. 

**Actual behavior**
InterruptIN tests do not pass. It seems that something about the InterruptIn object doesnt play nice with greentea on this platform. Possibly the destructor?

**Steps to reproduce**
Run `mbed test -n tests-api-interruptin -v` to see it fail. 
`mbed test -n tests-assumptions-pwmout` will pass, but if you uncomment the line about `x.disable_irq()' it will stop working

https://github.com/ARMmbed/ci-test-shield/blob/master/TESTS/API/InterruptIn/InterruptIn.cpp

BlackstoneEngineering, 2016-10-06:
@screamerbg

0xc0170, 2016-10-06:
cc @bcostm @adustm  @jeromecoutant @LMESTM

bridadan, 2016-10-06:
@adbridge Could interrupts be messing with utest at all?

adbridge, 2016-10-06:
Utest uses the us_ticker to schedule it's handlers (including test cases) once the next handler to run is identified it is passed to the main thread for execution. If there are strange things going on in the test cases themselves then all bets are off...

BlackstoneEngineering, 2016-10-06:
it works just find on K64F

adbridge, 2016-10-06:
mm Sounds more target related then...

LMESTM, 2016-10-06:
@jamike - I think he found out a new / recent issue with GCC init phases that may cause tickers to stop working - this might be related

BlackstoneEngineering, 2016-10-06:
Ran it with GCC too, failed as well

```
[1475772863.25][CONN][RXD] >>> Running case #1: 'InterruptIn on DIO_2'...
[1475772863.29][CONN][RXD] {{__testcase_start;InterruptIn on DIO_2}}
[1475772863.29][CONN][INF] found KV pair in stream: {{__testcase_start;InterruptIn on DIO_2}}, queued...
[1475772863.34][CONN][RXD] {{__testcase_finish;InterruptIn on DIO_2;1;0}}
[1475772863.34][CONN][INF] found KV pair in stream: {{__testcase_finish;InterruptIn on DIO_2;1;0}}, queued...
[1475772863.39][CONN][RXD] >>> 'InterruptIn on DIO_2': 1 passed, 0 failed
[1475772863.44][CONN][RXD] >>> Running case #2: 'InterruptIn on DIO_3'...
[1475772863.48][CONN][RXD] {{__testcase_start;InterruptIn on DIO_3}}
[1475772863.48][CONN][INF] found KV pair in stream: {{__testcase_start;InterruptIn on DIO_3}}, queued...
[1475772903.03][HTST][INF] test suite run finished after 40.24 sec...
[1475772903.03][CONN][INF] received special even '__host_test_finished' value='True', finishing
[1475772903.04][HTST][INF] CONN exited with code: 0
[1475772903.04][HTST][INF] No events in queue
[1475772903.04][HTST][INF] stopped consuming events
[1475772903.04][HTST][INF] host test result(): None
[1475772903.04][HTST][WRN] missing __exit event from DUT
[1475772903.04][HTST][WRN] missing __exit_event_queue event from host test
[1475772903.04][HTST][ERR] missing __exit_event_queue event from host test and no result from host test, timeout...
[1475772903.04][HTST][INF] calling blocking teardown()
[1475772903.04][HTST][INF] teardown() finished
[1475772903.04][HTST][INF] {{result;timeout}}
mbedgt: checking for GCOV data...
mbedgt: mbed-host-test-runner: stopped and returned 'TIMEOUT'
mbedgt: test on hardware with target id: 074002000765616A3800F717
mbedgt: test suite 'tests-api-interruptin' ........................................................... TIMEOUT in 49.03 sec
        test case: 'InterruptIn on DIO_2' ............................................................ OK in 0.05 sec
        test case: 'InterruptIn on DIO_3' ............................................................ ERROR in 0.00 sec
        test case: 'InterruptIn on DIO_4' ............................................................ SKIPPED in 0.00 sec
        test case: 'InterruptIn on DIO_5' ............................................................ SKIPPED in 0.00 sec
        test case: 'InterruptIn on DIO_6' ............................................................ SKIPPED in 0.00 sec
        test case: 'InterruptIn on DIO_7' ............................................................ SKIPPED in 0.00 sec
        test case: 'InterruptIn on DIO_8' ............................................................ SKIPPED in 0.00 sec
        test case: 'InterruptIn on DIO_9' ............................................................ SKIPPED in 0.00 sec
mbedgt: all tests finished!
mbedgt: shuffle seed: 0.9623053188
mbedgt: test suite report:
+-------------------+---------------+-----------------------+---------+--------------------+-------------+
| target            | platform_name | test suite            | result  | elapsed_time (sec) | copy_method |
+-------------------+---------------+-----------------------+---------+--------------------+-------------+
| NUCLEO_F411RE-ARM | NUCLEO_F411RE | tests-api-interruptin | TIMEOUT | 49.03              | shell       |
+-------------------+---------------+-----------------------+---------+--------------------+-------------+
mbedgt: test suite results: 1 TIMEOUT
mbedgt: test case report:
+-------------------+---------------+-----------------------+----------------------+--------+--------+---------+--------------------+
| target            | platform_name | test suite            | test case            | passed | failed | result  | elapsed_time (sec) |
+-------------------+---------------+-----------------------+----------------------+--------+--------+---------+--------------------+
| NUCLEO_F411RE-ARM | NUCLEO_F411RE | tests-api-interruptin | InterruptIn on DIO_2 | 1      | 0      | OK      | 0.05               |
| NUCLEO_F411RE-ARM | NUCLEO_F411RE | tests-api-interruptin | InterruptIn on DIO_3 | 0      | 0      | ERROR   | 0.0                |
| NUCLEO_F411RE-ARM | NUCLEO_F411RE | tests-api-interruptin | InterruptIn on DIO_4 | 0      | 0      | SKIPPED | 0.0                |
| NUCLEO_F411RE-ARM | NUCLEO_F411RE | tests-api-interruptin | InterruptIn on DIO_5 | 0      | 0      | SKIPPED | 0.0                |
| NUCLEO_F411RE-ARM | NUCLEO_F411RE | tests-api-interruptin | InterruptIn on DIO_6 | 0      | 0      | SKIPPED | 0.0                |
| NUCLEO_F411RE-ARM | NUCLEO_F411RE | tests-api-interruptin | InterruptIn on DIO_7 | 0      | 0      | SKIPPED | 0.0                |
| NUCLEO_F411RE-ARM | NUCLEO_F411RE | tests-api-interruptin | InterruptIn on DIO_8 | 0      | 0      | SKIPPED | 0.0                |
| NUCLEO_F411RE-ARM | NUCLEO_F411RE | tests-api-interruptin | InterruptIn on DIO_9 | 0      | 0      | SKIPPED | 0.0                |
+-------------------+---------------+-----------------------+----------------------+--------+--------+---------+--------------------+
mbedgt: test case results: 6 SKIPPED / 1 OK / 1 ERROR
mbedgt: completed in 49.58 sec
mbedgt: exited with code 1
```

bridadan, 2016-10-06:
@BlackstoneEngineering That log says it was using the `ARM` compiler

BlackstoneEngineering, 2016-10-06:
correct, armcc, they said it was just a gcc issue, so I tried it with armcc, both exhibit the behavior. Hence why I think its a platform issue.

LMESTM, 2016-10-07:
Ok thanks - so this is not related to GCC only ! 
Just to be clear, if we don't have the CI shield, running the test requires to connect 
D2 to D3, 
D4 to D5, 
D6 to D7
D8 to D9 
is it ?

BlackstoneEngineering, 2016-10-07:
For the interruptin tests yes, they are simple loopbacks.

adustm, 2016-11-21:
Hi,
I just pushed a fix for InterruptIn test. 
SYSCFG_EXTICR4,  EXTI_IMR, EXTI_RTSR, EXTI_FTSR registers were not in a correct state after the sw destruction of the InterruptIn object.
The next time those HW registers were activated by a new InterruptIn object, the interrupt management was all messed up.

adustm, 2016-12-16:
Hello, is it ok now ? Can we close this issue ?
Cheers

bridadan, 2016-12-16:
I'll go ahead and close this since #3303 was merged. @BlackstoneEngineering please reopen this if the issue persists.

closed 2017-01-19 by sg-