ARMmbed/mbed-os #2941: CFSTORE: cfstore_fsm_log_on_entry:Error: FlashJournal_commit() failed (ret=-1), possible cfstore_realloc_ex() error

simonqhughes, 2016-10-06:
Marcus Chang reports CFSTORE flush indications are not always being received. He supplied the 2 trace logs attached for investigation

simonqhughes, 2016-10-06:
These are the log files submitted.

[20161006_issue_2941_log.txt](https://github.com/ARMmbed/mbed-os/files/513499/20161006_issue_2941_log.txt)
[20161005_issue_2941_log.txt](https://github.com/ARMmbed/mbed-os/files/513500/20161005_issue_2941_log.txt)

simonqhughes, 2016-10-06:
There are 2 minor errors here:
- in the case where FlashJournal_commit() fails, the code path for the error handling doesnt generate the CFSTORE_OPCODE_FLUSH notification to the client.
- in the case where the cfstore is already initialized and initialize() is called again, the CFSTORE_OPCODE_INITIALIZE is not sent to the client.

sg-, 2017-01-19:
@simonqhughes Is this resolved?

closed 2017-10-27 by ChiefBureaucraticOfficer