ARMmbed/mbed-os #2944: cmsis update to v4

0xc0170, 2016-10-06:
### Description
- Type:  Question
- Priority: Major (might be considered a blocker for some future targets?)

## Question

 I quote here @akselsm from cmsis-core:
"The ARM mbed cmsis-core files are currently from CMSIS version 3.20, which is more than 3 years old by now. CMSIS version 4.5 was recently released, and the later versions of CMSIS define new macros for use by silicon vendors in device header files. This means that at some point in the future, silicon vendors may start to depend on symbols only defined in newer versions of CMSIS in their device header files.

Such symbols include the _I, _Oand _IO macros. In CMSIS 4.2 and newer, these have been replaced by _IM, _OM and _IOM for structure members. The old symbols still exist, but the M-suffixed ones are recommended for new devices.

How will ARM mbed support this going forward? Will the CMSIS core header files in cmsis-core be updated to newer versions, or will mbed require silicon vendors to maintain device header files compatible with CMSIS 3.20 for the foreseeable future? If the core headers are updated with backwards-incompatible changes, how will this be handled with regards to existing mbed enabled devices?"

cc @sg- @stevew817

akselsm, 2016-10-06:
CMSIS has been updated to version 4.00 at some point since the original ticket on the cmsis-core repo from December 2015, but that is still a too old version to have the _IOM macros, which were introduced in CMSIS 4.20. See e.g. https://github.com/ARMmbed/mbed-os/blob/master/targets/cmsis/core_cm4.h#L244 vs https://github.com/SiliconLabs/Gecko_SDK/blob/master/cmsis/Include/core_cm4.h#L274.

Our platform team has started generating device header files depending on the _IOM macros. This means that this is a real blocker for us being able to port new targets to mbed. Also, our hal libraries can't be updated either, since they also might depend on new device headers for bugfixes to register values/names etc.

sg-, 2017-01-19:
We're planning to update to CMSIS v5 in mbed OS 5.5

closed 2017-01-19 by sg-