ARMmbed/mbed-os #2945: CFSTORE: unexpected ARM_CFSTORE_DRIVER_ERROR_KEY_UNREADABLE error when finding KV

simonqhughes, 2016-10-06:
Brendan submitted the following log showing a find() returning ARM_CFSTORE_DRIVER_ERROR_KEY_UNREADABLE unexpectectly.

simonqhughes, 2016-10-06:
The log is attached here:

[20161006_issue_1012.txt](https://github.com/ARMmbed/mbed-os/files/513640/20161006_issue_1012.txt)

@bremoran please could you:
- provide a commit sha of the version of mbedos used for configstore
- provide a few lines of (pseudo-)code that I can turn into a test case to reproduce the problem (with the latest version of cfstore)

bremoran, 2016-10-06:
@simonqhughes I'm afraid that the test case that produces this error is quite involved. A few lines will not be possible without significant work.

For the moment, I will have to disable the test-case that is producing the error. I will work on a simplified test case when time permits.

mbed OS SHA1: dda7f7d77abd4330b05e686ce3bbe58230eb7876

closed 2017-10-27 by ChiefBureaucraticOfficer