ARMmbed/mbed-os #2950: Bug in K64F Linker script: __StackTop is set to invalid address

LiyouZhou, 2016-10-06:
in file `MK64FN1M0xxx12.ld`:
```
  /* Initializes stack on the end of block */
  __StackTop = ORIGIN(m_data_2) + LENGTH(m_data_2);
```
This sets __StackTop to an invalid address. Because __StackTop is ignored in mbed-os, the bug does not show up normally.

But if there is a bootloader, bootloader sets `SCB->VTOR` to the value of stack top before jumping to application. An invalid address would cause a Hardfault immediately before jumping.

0xc0170, 2016-10-06:
cc @mmahadevan108

LiyouZhou, 2016-10-06:
Actually I think because the bootloader calls `__set_MSP` with an invalid address

mmahadevan108, 2016-10-06:
__StackTop is set to the top of RAM (0x20030000 in case of K64F). This is as expected, I don't see the issue.

LiyouZhou, 2016-10-06:
@mmahadevan108 0x20030000 is not a valid memory address. The last valid memory address is 0x2002ffff.

mmahadevan108, 2016-10-06:
Across all mbed you will see the INITIAL_SP macro set to a similar value (0x20030000 in case of K64F). Martin, can you please clarify how this address is treated during stack operations?

c1728p9, 2016-10-06:
The stack pointer needs to be 8 bytes aligned, so setting it to 0x2002ffff will cause problems. Even though the address 0x20030000 is outside of ram the first push start below this address.

With a stack pointer of 0x20030000 a push operation will do this:
```
push {r0, r1}

r0 -> addr 2002FFFC 
r1 -> addr 2002FFF8
```
What kind of bootloader do you have? If the MSP isn't set to 0x20030000 by the bootloader this would cause problems for the application.

c1728p9, 2016-10-07:
@LiyouZhou can this issue be closed, or do you still have concerns?

mmahadevan108, 2016-12-14:
Can this issue be resolved?

closed 2016-12-14 by 0xc0170