ARMmbed/mbed-os #2955: UBLOX_EVK_ODIN_W2 - Pinout issues

screamerbg, 2016-10-06:
### Description
- Type: Bug
- Priority: Major

---------------------------------------------------------------
## Cannot set A0, A3 high (via DigitalOut)
Attach an LED and try it yourself

## Digital loop tests failures (DigitalIn + DigitalOut)
Digital I/O loop on D2 -> D3
Digital I/O loop on D3 -> D2
Digital I/O loop on D8 -> D9
Digital I/O loop on D9 -> D8

## Analog In failures (AnalogIn)
Analog Input on A1
Analog Input on A2
Analog Input on A3
Analog Input on A4 (suspecting that the app crashed)

## I2C failures with EEPROM (I2C)
I2C -  EEProm Read (fails with ARMCC, GCC)
I2C -  TMP102 Temperature Read (fails with GCC)

**Toolchain:**
GCC_ARM, ARM

**meed-os sha:**
22ec73b Merge pull request #2948 from jamike/Fixed_iarm_debug_built_failed

**Steps to reproduce**
Tests available here https://github.com/armmbed/ci-test-shield
Hardware components here https://github.com/ARMmbed/mbed-HDK/tree/master/Production%20Design%20Projects/CITestShield

CC @andreaslarssonublox

screamerbg, 2016-10-14:
I2C no longer an issue thanks to PR #2972. cc @adustm @bcostm @betzw @andreaslarssonublox

ramtin-r, 2016-10-26:
1. Only A1 and A2 can be as digital pins.

2. D2 is used for UART and can not be used for testing digital loop case.

3. A4 is disabled and can not be used at all.

4. SPI Failures -> Not clear what the problem is

cc @adustm @bcostm @betzw @andreaslarssonublox @screamerbg

bcostm, 2017-05-22:
Can you please remove the ST label and add Ublox label instead ?

0xc0170, 2017-05-25:
@andreaslarssonublox bump

andreaslarssonublox, 2017-05-29:
We ran the tests again with this SW:
mbed-os: SHA-1: 598cbdf023a4417a6af5cd4a5503b690722d5205
ci-shield: SHA-1: 691997cf918090584844c67f02d866df08b5fe5f

+---------------------------+-------------------+-----------------------+------------------------------------+--------+--------+--------+--------------------+
| target                    | platform_name     | test suite            | test case                          | passed | failed | result | elapsed_time (sec) |
+---------------------------+-------------------+-----------------------+------------------------------------+--------+--------+--------+--------------------+
| UBLOX_EVK_ODIN_W2-GCC_ARM | UBLOX_EVK_ODIN_W2 | tests-api-analogin    | Analog Input on AIN_0              | 0      | 2      | FAIL   | 0.27               |
| UBLOX_EVK_ODIN_W2-GCC_ARM | UBLOX_EVK_ODIN_W2 | tests-api-analogin    | Analog Input on AIN_1              | 0      | 4      | FAIL   | 0.49               |
| UBLOX_EVK_ODIN_W2-GCC_ARM | UBLOX_EVK_ODIN_W2 | tests-api-analogin    | Analog Input on AIN_2              | 0      | 1      | FAIL   | 0.17               |
| UBLOX_EVK_ODIN_W2-GCC_ARM | UBLOX_EVK_ODIN_W2 | tests-api-analogin    | Analog Input on AIN_3              | 0      | 3      | FAIL   | 0.37               |
| UBLOX_EVK_ODIN_W2-GCC_ARM | UBLOX_EVK_ODIN_W2 | tests-api-analogin    | Analog Input on AIN_4              | 0      | 0      | ERROR  | 0.0                |
| UBLOX_EVK_ODIN_W2-GCC_ARM | UBLOX_EVK_ODIN_W2 | tests-api-businout    | tests-api-businout                 | 0      | 1      | ERROR  | 48.46              |
| UBLOX_EVK_ODIN_W2-GCC_ARM | UBLOX_EVK_ODIN_W2 | tests-api-digitalio   | Digital I/O on DIO_2/DIO_3         | 0      | 1      | FAIL   | 0.12               |
| UBLOX_EVK_ODIN_W2-GCC_ARM | UBLOX_EVK_ODIN_W2 | tests-api-digitalio   | Digital I/O on DIO_3/DIO_2         | 0      | 1      | FAIL   | 0.12               |
| UBLOX_EVK_ODIN_W2-GCC_ARM | UBLOX_EVK_ODIN_W2 | tests-api-digitalio   | Digital I/O on DIO_4/DIO_5         | 1      | 0      | OK     | 0.06               |
| UBLOX_EVK_ODIN_W2-GCC_ARM | UBLOX_EVK_ODIN_W2 | tests-api-digitalio   | Digital I/O on DIO_5/DIO_4         | 1      | 0      | OK     | 0.06               |
| UBLOX_EVK_ODIN_W2-GCC_ARM | UBLOX_EVK_ODIN_W2 | tests-api-digitalio   | Digital I/O on DIO_6/DIO_7         | 1      | 0      | OK     | 0.05               |
| UBLOX_EVK_ODIN_W2-GCC_ARM | UBLOX_EVK_ODIN_W2 | tests-api-digitalio   | Digital I/O on DIO_7/DIO_6         | 1      | 0      | OK     | 0.05               |
| UBLOX_EVK_ODIN_W2-GCC_ARM | UBLOX_EVK_ODIN_W2 | tests-api-i2c         | I2C -  EEProm 2nd WR 10  Bytes     | 1      | 0      | OK     | 0.06               |
| UBLOX_EVK_ODIN_W2-GCC_ARM | UBLOX_EVK_ODIN_W2 | tests-api-i2c         | I2C -  EEProm 2nd WR 100 Bytes     | 1      | 0      | OK     | 0.08               |
| UBLOX_EVK_ODIN_W2-GCC_ARM | UBLOX_EVK_ODIN_W2 | tests-api-i2c         | I2C -  EEProm 2nd WR 2 Bytes       | 1      | 0      | OK     | 0.06               |
| UBLOX_EVK_ODIN_W2-GCC_ARM | UBLOX_EVK_ODIN_W2 | tests-api-i2c         | I2C -  EEProm 2nd WR Single Byte   | 1      | 0      | OK     | 0.06               |
| UBLOX_EVK_ODIN_W2-GCC_ARM | UBLOX_EVK_ODIN_W2 | tests-api-i2c         | I2C -  EEProm WR 10  Bytes         | 1      | 0      | OK     | 0.06               |
| UBLOX_EVK_ODIN_W2-GCC_ARM | UBLOX_EVK_ODIN_W2 | tests-api-i2c         | I2C -  EEProm WR 100 Bytes         | 1      | 0      | OK     | 0.07               |
| UBLOX_EVK_ODIN_W2-GCC_ARM | UBLOX_EVK_ODIN_W2 | tests-api-i2c         | I2C -  EEProm WR 2 Bytes           | 1      | 0      | OK     | 0.05               |
| UBLOX_EVK_ODIN_W2-GCC_ARM | UBLOX_EVK_ODIN_W2 | tests-api-i2c         | I2C -  EEProm WR Single Byte       | 1      | 0      | OK     | 0.06               |
| UBLOX_EVK_ODIN_W2-GCC_ARM | UBLOX_EVK_ODIN_W2 | tests-api-i2c         | I2C -  Instantiation of I2C Object | 1      | 0      | OK     | 0.06               |
| UBLOX_EVK_ODIN_W2-GCC_ARM | UBLOX_EVK_ODIN_W2 | tests-api-i2c         | I2C -  LM75B Temperature Read      | 1      | 0      | OK     | 0.06               |
| UBLOX_EVK_ODIN_W2-GCC_ARM | UBLOX_EVK_ODIN_W2 | tests-api-interruptin | InterruptIn on DIO_2               | 0      | 2      | FAIL   | 0.16               |
| UBLOX_EVK_ODIN_W2-GCC_ARM | UBLOX_EVK_ODIN_W2 | tests-api-interruptin | InterruptIn on DIO_3               | 0      | 2      | FAIL   | 0.17               |
| UBLOX_EVK_ODIN_W2-GCC_ARM | UBLOX_EVK_ODIN_W2 | tests-api-interruptin | InterruptIn on DIO_4               | 1      | 0      | OK     | 0.06               |
| UBLOX_EVK_ODIN_W2-GCC_ARM | UBLOX_EVK_ODIN_W2 | tests-api-interruptin | InterruptIn on DIO_5               | 1      | 0      | OK     | 0.05               |
| UBLOX_EVK_ODIN_W2-GCC_ARM | UBLOX_EVK_ODIN_W2 | tests-api-interruptin | InterruptIn on DIO_6               | 1      | 0      | OK     | 0.04               |
| UBLOX_EVK_ODIN_W2-GCC_ARM | UBLOX_EVK_ODIN_W2 | tests-api-interruptin | InterruptIn on DIO_7               | 1      | 0      | OK     | 0.05               |
| UBLOX_EVK_ODIN_W2-GCC_ARM | UBLOX_EVK_ODIN_W2 | tests-api-pwm         | Pwm object definable               | 0      | 0      | ERROR  | 0.0                |
| UBLOX_EVK_ODIN_W2-GCC_ARM | UBLOX_EVK_ODIN_W2 | tests-api-spi         | SPI - Object Definable             | 1      | 0      | OK     | 0.05               |
[... cut ...]
- The tests-api-analogin won't pass because A0 and A1 are configured as either input or output by changing the jumpers J26.
- The tests-api-digitalio won't pass since D2 is used as debug UART RX. If this pin can be omitted from the tests it would pass.
- The tests-api-interruptin won't pass see tests-api-digitalio.
- The tests-api-pwm won't pass since we don't support PWM on all pins. If some pins can be omitted from the test it would pass.
- We have created an internal PR for tests-api-spi, TE_OM2-58

@0xc0170 Is it mandatory to pass those tests for mbed-os 5.5?

0xc0170, 2017-05-30:
> @0xc0170 Is it mandatory to pass those tests for mbed-os 5.5?

Not yet. The ci test shield repo will go through some updates to address some of the comments above regarding not all pins are available for all peripherals.

Do we need to update https://github.com/ARMmbed/ci-test-shield/blob/master/mbed_app.json to reflect any of these failures (update pins) ?

andreaslarssonublox, 2017-05-31:
Ok.
I don't think we can fix anything by updating the mbed_app.json since it will only replace pins I guess.

0xc0170, 2018-06-05:
@andreaslarssonublox Is this as won't fix or already resolved? It's been some time since the last update.

deepikabhavnani, 2018-09-11:
Mandatory ?
> Not yet. The ci test shield repo will go through some updates to address some of the comments above regarding not all pins are available for all peripherals.
> > I don't think we will fix it unless mandatory

Closing it, please reopen later

closed 2018-09-11 by deepikabhavnani