ARMmbed/mbed-os #2956: NUCLEO_F429ZI Issues

screamerbg, 2016-10-06:
Note: This is just a template, so feel free to use/remove the unnecessary things

### Description
- Type: Bug
- Priority: Major

---------------------------------------------------------------
## Analog In failures (AnalogIn)
Analog Input on A0 (fails with GCC, but works with ARMCC)

## I2C failures with EEPROM (I2C)
I2C -  EEProm Read (fails with ARMCC, GCC)
I2C -  TMP102 Temperature Read (fails with GCC, works with ARMCC)

**Target**
NUCLEO_F429ZI

**Toolchain:**
GCC_ARM+ARM

**meed-os sha:**
22ec73b Merge pull request #2948 from jamike/Fixed_iarm_debug_built_failed

**Steps to reproduce**
Tests available here https://github.com/armmbed/ci-test-shield
Hardware components here https://github.com/ARMmbed/mbed-HDK/tree/master/Production%20Design%20Projects/CITestShield

CC @adustm @bcostm

bcostm, 2016-10-13:
I'll take a look.

screamerbg, 2016-10-14:
I2C no longer an issue thanks to PR #2972

bcostm, 2016-10-17:
Some news:

*InterruptIn failures*
- Confirmed on D3 with ARM and GCC_ARM toolchains.
- Debug on-going. I need to create first a uVision project because project.py does not work anymore...

*AnalogIn failures*
- Confirmed on A0 with GCC_ARM.
- OK with ARM.
- Issue solved by increasing the ADC sampling time to 28 cycles (instead of 15). A pull Request will be done.

0xc0170, 2016-10-17:
@bcostm Thanks for looking at these issues

bcostm, 2016-10-17:
For the AnalogIn, the error comes from the test itself. The previous_value is initialized to 0 and sometimes the first convertion is also equal to 0... And the test is fail due to the assert who is checking that the current_value > previous_value...

I kept the initial 15 sampling time, and after correcting the test:
```
+-------------------+---------------+--------------------+-----------------------+--------+--------+--------+--------------------+
| target            | platform_name | test suite         | test case             | passed | failed | result | elapsed_time (sec) |
+-------------------+---------------+--------------------+-----------------------+--------+--------+--------+--------------------+
| NUCLEO_F429ZI-ARM | NUCLEO_F429ZI | tests-api-analogin | Analog Input on AIN_0 | 1      | 0      | OK     | 0.32               |
| NUCLEO_F429ZI-ARM | NUCLEO_F429ZI | tests-api-analogin | Analog Input on AIN_1 | 1      | 0      | OK     | 0.33               |
| NUCLEO_F429ZI-ARM | NUCLEO_F429ZI | tests-api-analogin | Analog Input on AIN_2 | 1      | 0      | OK     | 0.36               |
| NUCLEO_F429ZI-ARM | NUCLEO_F429ZI | tests-api-analogin | Analog Input on AIN_3 | 1      | 0      | OK     | 0.31               |
| NUCLEO_F429ZI-ARM | NUCLEO_F429ZI | tests-api-analogin | Analog Input on AIN_4 | 1      | 0      | OK     | 0.33               |
| NUCLEO_F429ZI-ARM | NUCLEO_F429ZI | tests-api-analogin | Analog Input on AIN_5 | 1      | 0      | OK     | 0.33               |
+-------------------+---------------+--------------------+-----------------------+--------+--------+--------+--------------------+

+-----------------------+---------------+--------------------+-----------------------+--------+--------+--------+--------------------+
| target                | platform_name | test suite         | test case             | passed | failed | result | elapsed_time (sec) |
+-----------------------+---------------+--------------------+-----------------------+--------+--------+--------+--------------------+
| NUCLEO_F429ZI-GCC_ARM | NUCLEO_F429ZI | tests-api-analogin | Analog Input on AIN_0 | 1      | 0      | OK     | 0.32               |
| NUCLEO_F429ZI-GCC_ARM | NUCLEO_F429ZI | tests-api-analogin | Analog Input on AIN_1 | 1      | 0      | OK     | 0.34               |
| NUCLEO_F429ZI-GCC_ARM | NUCLEO_F429ZI | tests-api-analogin | Analog Input on AIN_2 | 1      | 0      | OK     | 0.36               |
| NUCLEO_F429ZI-GCC_ARM | NUCLEO_F429ZI | tests-api-analogin | Analog Input on AIN_3 | 1      | 0      | OK     | 0.35               |
| NUCLEO_F429ZI-GCC_ARM | NUCLEO_F429ZI | tests-api-analogin | Analog Input on AIN_4 | 1      | 0      | OK     | 0.35               |
| NUCLEO_F429ZI-GCC_ARM | NUCLEO_F429ZI | tests-api-analogin | Analog Input on AIN_5 | 1      | 0      | OK     | 0.32               |
+-----------------------+---------------+--------------------+-----------------------+--------+--------+--------+--------------------+

```

screamerbg, 2016-10-17:
@blackstoneengineering

BlackstoneEngineering, 2016-10-17:
@bcostm how did you correct the test? Mind providing code or submitting a PR against the CI Test shield?

I assume you just initialized `outputs` to 1?

bcostm, 2016-10-18:
Concerning the InterruptIn, I understood the problem but it is not solved. It is because the same interrupt vector is used for GPIO pins Px_5 to Px_9 and another one for pins Px_10 to Px_15.
I have tested each pin individually and they are all ok.
If you test for example D6 (pin Px_9) and any other pin (Px_10 to Px_15) it is ok. But as soon as two pins are tested which are on the same IT vector the test crashes...

bcostm, 2016-11-14:
I have created a uvision project using "mbed export" to debug the InterruptIn issue on Nucleo_F429ZI.
 
But the program is waiting for something on the serial port (continuous loop in serial_readable function) and the tests are not executed.

Someone has an idea how to debug these ci tests ? Is there something to enable ?

BlackstoneEngineering, 2016-11-14:
@bcostm are you using Uvision 4 or 5?
Also, greentea tests are not meant to be run from within uvision, they're meant to be run with greentea, ie the `mbed test` command, so if you just run the test in uvision you wont have the other half of the test ecosystem running (the greentea bit on the computer that catches and drives the tests), so I would expect it to hang infinitely.

bcostm, 2016-11-14:
I use uvision 5 and this is why I thought. So, it's very difficult to debug this test case :(

FYI I tried also on the NUCLEO_F411RE. Same fail on D3 pin (=PB_3) when it follows the test of D2 pin (PA_10). But in this case they are not using the same interrupt vector as on the F429. So the idea I add previously is not the good one unfortunately, there is something else...

adustm, 2016-11-15:
Hello @BlackstoneEngineering , @0xc0170 , @screamerbg 
Do you have any clue to use a debugger together with greentea ? We need it in order to be able to read HW registers etc... in case of test failure. 
Cheers

bridadan, 2016-11-15:
Hi @adustm, I have some instructions for using a debugger with greentea here: https://github.com/ARMmbed/mbed-os/blob/master/docs/testing_mbed_OS_5.md#debugging-tests

Please let me know if you have any issues!

bcostm, 2016-11-16:
Hi,
Thanks for this information. I can now debug the test :+1:

bridadan, 2016-11-16:
Where are you setting the breakpoint? And perhaps double check the debugger settings on uVision?

bridadan, 2016-11-16:
Also another thought, do you know if its possible to use the debugger while also using the serial port on STLink devices? I know there are other interface firmwares out there that don't allow this (I think CMSIS-DAP was one of them)

adustm, 2016-11-18:
Hello @bridadan 
It works fine also for me, in another context using GCC_ARM.
Thanks
Armelle

adustm, 2016-11-21:
Hi,
I just pushed a fix for InterruptIn test. 
SYSCFG_EXTICRx,  EXTI_IMR, EXTI_RTSR, EXTI_FTSR registers were not in a correct state after the sw destruction of the InterruptIn object.
The next time those HW registers were activated by a new InterruptIn object, the interrupt management was all messed up.

commit 0219b64a (2016-12-02): fix #2956. Add HAL_DeInit function if gpio_irq destructor
This allows ci-test-shield tests-api-interruptin to pass
commit 1c2c1217 (2016-12-02): Merge pull request #3303 from adustm/stm_fix_interrupt_in

Fix #2956 #2939 #2957 #2959 #2960: Add HAL_DeInit function in gpio_irq destructor
closed 2016-12-02 by 0xc0170