ARMmbed/mbed-os #2957: NUCLEO_L476RG Issues

screamerbg, 2016-10-06:
### Description
- Type: Bug
- Priority: Major

---------------------------------------------------------------

## Interrupt failures (InterruptIn) - GCC and ARMCC
InterruptIn on D2 actually works
InterruptIn on D3 throws greentea error, I suspect that the test crashes
InterruptIn on D4-9 are skipped due to the error with InterruptIn on D3

## Analog In failures (AnalogIn) - GCC and ARMCC
Analog Input on A0
Analog Input on A1
Analog Input on A2
Analog Input on A3
Analog Input on A4
Analog Input on A5

**Target**
NUCLEO_L476RG

**Toolchain:**
GCC_ARM, ARM

**meed-os sha:**
22ec73b Merge pull request #2948 from jamike/Fixed_iarm_debug_built_failed

**Steps to reproduce**
Tests available here https://github.com/armmbed/ci-test-shield
Hardware components here https://github.com/ARMmbed/mbed-HDK/tree/master/Production%20Design%20Projects/CITestShield

CC @adustm @bcostm

BlackstoneEngineering, 2016-10-07:
#ci-test-shield for the win!

adustm, 2016-11-21:
Hi,
I just pushed a fix for InterruptIn test. 
SYSCFG_EXTICR4,  EXTI_IMR, EXTI_RTSR, EXTI_FTSR registers were not in a correct state after the sw destruction of the InterruptIn object.
The next time those HW registers were activated by a new InterruptIn object, the interrupt management was all messed up.

bcostm, 2017-03-03:
Retested with mbed-os-5.4 and InterruptIn + Analog tests are OK:

```
+-----------------------+---------------+-----------------------+------------------------------------+--------+--------+--------+--------------------+
| target                | platform_name | test suite            | test case                          | passed | failed | result | elapsed_time (sec) |
+-----------------------+---------------+-----------------------+------------------------------------+--------+--------+--------+--------------------+
| NUCLEO_L476RG-GCC_ARM | NUCLEO_L476RG | tests-api-analogin    | Analog Input on AIN_0              | 1      | 0      | OK     | 0.04               |
| NUCLEO_L476RG-GCC_ARM | NUCLEO_L476RG | tests-api-analogin    | Analog Input on AIN_1              | 1      | 0      | OK     | 0.07               |
| NUCLEO_L476RG-GCC_ARM | NUCLEO_L476RG | tests-api-analogin    | Analog Input on AIN_2              | 1      | 0      | OK     | 0.05               |
| NUCLEO_L476RG-GCC_ARM | NUCLEO_L476RG | tests-api-analogin    | Analog Input on AIN_3              | 1      | 0      | OK     | 0.06               |
| NUCLEO_L476RG-GCC_ARM | NUCLEO_L476RG | tests-api-analogin    | Analog Input on AIN_4              | 1      | 0      | OK     | 0.07               |
| NUCLEO_L476RG-GCC_ARM | NUCLEO_L476RG | tests-api-analogin    | Analog Input on AIN_5              | 1      | 0      | OK     | 0.05               |
| NUCLEO_L476RG-GCC_ARM | NUCLEO_L476RG | tests-api-analogout   | AnalogOut on AOUT                  | 1      | 0      | OK     | 0.05               |
| NUCLEO_L476RG-GCC_ARM | NUCLEO_L476RG | tests-api-businout    | BusIn definable                    | 1      | 0      | OK     | 0.05               |
| NUCLEO_L476RG-GCC_ARM | NUCLEO_L476RG | tests-api-businout    | BusIn to BusOut test               | 1      | 0      | OK     | 0.06               |
| NUCLEO_L476RG-GCC_ARM | NUCLEO_L476RG | tests-api-businout    | BusInOut to BusInOut               | 1      | 0      | OK     | 16.03              |
| NUCLEO_L476RG-GCC_ARM | NUCLEO_L476RG | tests-api-businout    | BusOut definable                   | 1      | 0      | OK     | 0.05               |
| NUCLEO_L476RG-GCC_ARM | NUCLEO_L476RG | tests-api-digitalio   | Digital I/O on DIO_2/DIO_3         | 1      | 0      | OK     | 0.06               |
| NUCLEO_L476RG-GCC_ARM | NUCLEO_L476RG | tests-api-digitalio   | Digital I/O on DIO_3/DIO_2         | 1      | 0      | OK     | 0.06               |
| NUCLEO_L476RG-GCC_ARM | NUCLEO_L476RG | tests-api-digitalio   | Digital I/O on DIO_4/DIO_5         | 1      | 0      | OK     | 0.04               |
| NUCLEO_L476RG-GCC_ARM | NUCLEO_L476RG | tests-api-digitalio   | Digital I/O on DIO_5/DIO_4         | 1      | 0      | OK     | 0.06               |
| NUCLEO_L476RG-GCC_ARM | NUCLEO_L476RG | tests-api-digitalio   | Digital I/O on DIO_6/DIO_7         | 1      | 0      | OK     | 0.05               |
| NUCLEO_L476RG-GCC_ARM | NUCLEO_L476RG | tests-api-digitalio   | Digital I/O on DIO_7/DIO_6         | 1      | 0      | OK     | 0.06               |
| NUCLEO_L476RG-GCC_ARM | NUCLEO_L476RG | tests-api-digitalio   | Digital I/O on DIO_8/DIO_9         | 1      | 0      | OK     | 0.05               |
| NUCLEO_L476RG-GCC_ARM | NUCLEO_L476RG | tests-api-digitalio   | Digital I/O on DIO_9/DIO_8         | 1      | 0      | OK     | 0.06               |
| NUCLEO_L476RG-GCC_ARM | NUCLEO_L476RG | tests-api-i2c         | I2C -  EEProm 2nd WR 10  Bytes     | 1      | 0      | OK     | 0.06               |
| NUCLEO_L476RG-GCC_ARM | NUCLEO_L476RG | tests-api-i2c         | I2C -  EEProm 2nd WR 100 Bytes     | 1      | 0      | OK     | 0.14               |
| NUCLEO_L476RG-GCC_ARM | NUCLEO_L476RG | tests-api-i2c         | I2C -  EEProm 2nd WR 2 Bytes       | 1      | 0      | OK     | 0.06               |
| NUCLEO_L476RG-GCC_ARM | NUCLEO_L476RG | tests-api-i2c         | I2C -  EEProm 2nd WR Single Byte   | 1      | 0      | OK     | 0.09               |
| NUCLEO_L476RG-GCC_ARM | NUCLEO_L476RG | tests-api-i2c         | I2C -  EEProm WR 10  Bytes         | 1      | 0      | OK     | 0.04               |
| NUCLEO_L476RG-GCC_ARM | NUCLEO_L476RG | tests-api-i2c         | I2C -  EEProm WR 100 Bytes         | 1      | 0      | OK     | 0.11               |
| NUCLEO_L476RG-GCC_ARM | NUCLEO_L476RG | tests-api-i2c         | I2C -  EEProm WR 2 Bytes           | 0      | 3      | FAIL   | 0.22               |
| NUCLEO_L476RG-GCC_ARM | NUCLEO_L476RG | tests-api-i2c         | I2C -  EEProm WR Single Byte       | 1      | 0      | OK     | 0.07               |
| NUCLEO_L476RG-GCC_ARM | NUCLEO_L476RG | tests-api-i2c         | I2C -  Instantiation of I2C Object | 1      | 0      | OK     | 0.07               |
| NUCLEO_L476RG-GCC_ARM | NUCLEO_L476RG | tests-api-i2c         | I2C -  LM75B Temperature Read      | 1      | 0      | OK     | 0.06               |
| NUCLEO_L476RG-GCC_ARM | NUCLEO_L476RG | tests-api-interruptin | InterruptIn on DIO_2               | 1      | 0      | OK     | 0.07               |
| NUCLEO_L476RG-GCC_ARM | NUCLEO_L476RG | tests-api-interruptin | InterruptIn on DIO_3               | 1      | 0      | OK     | 0.04               |
| NUCLEO_L476RG-GCC_ARM | NUCLEO_L476RG | tests-api-interruptin | InterruptIn on DIO_4               | 1      | 0      | OK     | 0.1                |
[... cut ...]
| NUCLEO_L476RG-GCC_ARM | NUCLEO_L476RG | tests-api-pwm         | PWM_3 Frequency 30ms               | 1      | 0      | OK     | 3.04               |
| NUCLEO_L476RG-GCC_ARM | NUCLEO_L476RG | tests-api-pwm         | Pwm object definable               | 1      | 0      | OK     | 0.08               |
+-----------------------+---------------+-----------------------+------------------------------------+--------+--------+--------+--------------------+
mbedgt: test case results: 1 FAIL / 61 OK
```

Would you mind closing this issue ?

I will open a new one concerning the FAIL with I2C/EEP test.

screamerbg, 2017-03-03:
This is great @bcostm! Thanks for taking care of these! :+1:

closed 2017-03-03 by screamerbg