ARMmbed/mbed-os #2959: NUCLEO_F446RE Issues

screamerbg, 2016-10-06:
Note: This is just a template, so feel free to use/remove the unnecessary things

### Description
- Type: Bug
- Priority: Major

---------------------------------------------------------------
## Pinout issue
There is a strange issue with the pins on this board.
It's better explained as comparison between F446 and L476, e.g.

ST Nucleo L476 (correct sequence)
https://www.youtube.com/watch?v=RhlTQsdUkR8

ST Nucleo F446 (incorrect sequence)
https://www.youtube.com/watch?v=aWWTfSxIlDE

It seems that LEDs on D1, D2, D4, D5, D7, D8 cannot be high, though D0, D3, D6, D9 function normally. This is mostly noticeable in the first sequence when LEDs are turned on in triplets, e.g. D0+D1+D2, D3+D4+D5, D6+D7+D8, etc.

The example program can be found here
https://developer.mbed.org/teams/shields/code/mbed_ProtoLED_Shield/file/b760013d9d08/main.cpp

## Interrupt failures (InterruptIn)
InterruptIn on D2 throws greentea error, I suspect that the test crashes
InterruptIn on D3-9 are skipped due to the error with InterruptIn on D3
(Note that unlike the other reports, no InterruptIn works)

## I2C failures with EEPROM (I2C)
I2C -  EEProm Read (fails with ARMCC, GCC)
I2C -  TMP102 Temperature Read (fails with GCC, works with ARMCC)

**Target**
NUCLEO_F446RE

**Toolchain:**
GCC_ARM, ARM

**meed-os sha:**
22ec73b Merge pull request #2948 from jamike/Fixed_iarm_debug_built_failed

**Steps to reproduce**
Tests available here https://github.com/armmbed/ci-test-shield
Hardware components here https://github.com/ARMmbed/mbed-HDK/tree/master/Production%20Design%20Projects/CITestShield

CC @adustm @bcostm

BlackstoneEngineering, 2016-10-07:
#ci-test-shield for the win!

screamerbg, 2016-10-14:
I2C no longer an issue thanks to PR #2972. cc @adustm @bcostm @betzw

adustm, 2016-11-21:
Hi,
I just pushed a fix for InterruptIn test. 
SYSCFG_EXTICR4,  EXTI_IMR, EXTI_RTSR, EXTI_FTSR registers were not in a correct state after the sw destruction of the InterruptIn object.
The next time those HW registers were activated by a new InterruptIn object, the interrupt management was all messed up.

bcostm, 2017-03-03:
Re-checked IOs with mbed-os-5.4 and ci-test shield:

```
| NUCLEO_F446RE-GCC_ARM | NUCLEO_F446RE | tests-api-digitalio   | Digital I/O on DIO_2/DIO_3         | 1      | 0      | OK     | 0.06               |
| NUCLEO_F446RE-GCC_ARM | NUCLEO_F446RE | tests-api-digitalio   | Digital I/O on DIO_3/DIO_2         | 1      | 0      | OK     | 0.07               |
| NUCLEO_F446RE-GCC_ARM | NUCLEO_F446RE | tests-api-digitalio   | Digital I/O on DIO_4/DIO_5         | 1      | 0      | OK     | 0.06               |
| NUCLEO_F446RE-GCC_ARM | NUCLEO_F446RE | tests-api-digitalio   | Digital I/O on DIO_5/DIO_4         | 1      | 0      | OK     | 0.05               |
| NUCLEO_F446RE-GCC_ARM | NUCLEO_F446RE | tests-api-digitalio   | Digital I/O on DIO_6/DIO_7         | 1      | 0      | OK     | 0.07               |
| NUCLEO_F446RE-GCC_ARM | NUCLEO_F446RE | tests-api-digitalio   | Digital I/O on DIO_7/DIO_6         | 1      | 0      | OK     | 0.05               |
| NUCLEO_F446RE-GCC_ARM | NUCLEO_F446RE | tests-api-digitalio   | Digital I/O on DIO_8/DIO_9         | 1      | 0      | OK     | 0.04               |
| NUCLEO_F446RE-GCC_ARM | NUCLEO_F446RE | tests-api-digitalio   | Digital I/O on DIO_9/DIO_8         | 1      | 0      | OK     | 0.05               |

| NUCLEO_F446RE-GCC_ARM | NUCLEO_F446RE | tests-api-interruptin | InterruptIn on DIO_2               | 1      | 0      | OK     | 0.05               |
| NUCLEO_F446RE-GCC_ARM | NUCLEO_F446RE | tests-api-interruptin | InterruptIn on DIO_3               | 1      | 0      | OK     | 0.05               |
| NUCLEO_F446RE-GCC_ARM | NUCLEO_F446RE | tests-api-interruptin | InterruptIn on DIO_4               | 1      | 0      | OK     | 0.04               |
| NUCLEO_F446RE-GCC_ARM | NUCLEO_F446RE | tests-api-interruptin | InterruptIn on DIO_5               | 1      | 0      | OK     | 0.05               |
| NUCLEO_F446RE-GCC_ARM | NUCLEO_F446RE | tests-api-interruptin | InterruptIn on DIO_6               | 0      | 0      | ERROR  | 0.0                |

```

Don't know why there is this fail on InterruptIn on D6 ? To be analyzed.

For information checked on NUCLEO_F446ZE and it is OK:
```
| NUCLEO_F446ZE-GCC_ARM | NUCLEO_F446ZE | tests-api-interruptin | InterruptIn on DIO_2               | 1      | 0      | OK     | 0.06               |
| NUCLEO_F446ZE-GCC_ARM | NUCLEO_F446ZE | tests-api-interruptin | InterruptIn on DIO_3               | 1      | 0      | OK     | 0.05               |
| NUCLEO_F446ZE-GCC_ARM | NUCLEO_F446ZE | tests-api-interruptin | InterruptIn on DIO_4               | 1      | 0      | OK     | 0.05               |
| NUCLEO_F446ZE-GCC_ARM | NUCLEO_F446ZE | tests-api-interruptin | InterruptIn on DIO_5               | 1      | 0      | OK     | 0.04               |
| NUCLEO_F446ZE-GCC_ARM | NUCLEO_F446ZE | tests-api-interruptin | InterruptIn on DIO_6               | 1      | 0      | OK     | 0.04               |
| NUCLEO_F446ZE-GCC_ARM | NUCLEO_F446ZE | tests-api-interruptin | InterruptIn on DIO_7               | 1      | 0      | OK     | 0.07               |
| NUCLEO_F446ZE-GCC_ARM | NUCLEO_F446ZE | tests-api-interruptin | InterruptIn on DIO_8               | 1      | 0      | OK     | 0.05               |
| NUCLEO_F446ZE-GCC_ARM | NUCLEO_F446ZE | tests-api-interruptin | InterruptIn on DIO_9               | 1      | 0      | OK     | 0.05               |
```

LMESTM, 2017-06-30:
@screamerbg @bcostm @adustm  Fix for F446RE (and others) proposed in #4676.

@screamerbg  would be great if you could run your LEDs demo again on F446RE with this patch included and check this is ok.

LMESTM, 2017-08-17:
#4676 has been merged, if no other error is reported, I propose to close this issue.
ST_TO_BE_CLOSED

closed 2017-08-17 by 0xc0170