ARMmbed/mbed-os #2960: NUCLEO_L746ZG Issues

screamerbg, 2016-10-06:
### Description
- Type: Bug
- Priority: Major

---------------------------------------------------------------

## Interrupt failures (InterruptIn)
InterruptIn on D2 actually works
InterruptIn on D3 throws greentea error, I suspect that the test crashes
InterruptIn on D4-9 are skipped due to the error with InterruptIn on D3

## Analog In failures (AnalogIn)
Analog Input on A0 (fails with ARMCC, works with GCC)
Analog Input on A5 (fails with GCC, works with ARMCC)

**Target**
NUCLEO_F746ZG

**Toolchain:**
GCC_ARM+ARM

**meed-os sha:**
22ec73b Merge pull request #2948 from jamike/Fixed_iarm_debug_built_failed

**Steps to reproduce**
Tests available here https://github.com/armmbed/ci-test-shield
Hardware components here https://github.com/ARMmbed/mbed-HDK/tree/master/Production%20Design%20Projects/CITestShield

CC @adustm @bcostm

BlackstoneEngineering, 2016-10-07:
#ci-test-shieldfor the win!

screamerbg, 2016-10-14:
Added I2C issues @bcostm @adustm

screamerbg, 2016-10-14:
Note that ST Discovery F746NG based on the same silicon has issues with AnalogIn A0 only, not A5. A clue that this might be peripheral configuration issue

jeromecoutant, 2017-05-11:
Hi @screamerbg 

All tests with ci-shield are OK with NUCLEO_746ZG

        "NUCLEO_F746ZG": {
             "AOUT": "D13"
        },

Can you close the issue ?

bcostm, 2017-05-18:
ST_TO_BE_CLOSED

LMESTM, 2017-06-30:
@sg- @screamerbg @0xc0170  please close as all the CI tests are now passed on this board

closed 2017-06-30 by 0xc0170