ARMmbed/mbed-os #2961: Version of mbed-os in online IDE is wrong

infinnovation, 2016-10-07:
### Description
- Type: Bug
- Priority: Major?

---------------------------------------------------------------
## Bug

**meed-os sha:**
670b098 Merge pull request #2917 from c1728p9/fix_InitTCs

**Expected behavior**
Program imported from online IDE gets same version of mbed-os

**Actual behavior**
Imported version is 670b098; online is some older version

**Steps to reproduce**
Create program online using new 2-argument version of TCPServer::accept
```c++
...
TCPSocket sock;
SocketAddress addr;
srv.accept(&sock, &addr);
```

Add mbed-os library, which shows as synced to revision 2540:670b098.
Try to compile.  It fails:
```
Error: Too many arguments in function call in "main.cpp", Line: 177, Col: 38
```

Now import to your PC with "mbed-cli import ..."
"mbed compile" succeeds!

screamerbg, 2016-10-07:
Note that the mbed Online IDE supports only the official (tagged) releases of mbed OS. If non-tagged revision is used then the mbed Online IDE uses the latest mbed OS release (which might be different to master).

infinnovation, 2016-10-07:
OK, easier example:

New program -> Template: mbed OS Blinky LED HelloWorld
In Program Workspace, right-click on mbed-os -> Update...
The Library Details pane now shows Revision 2540:670b098

This is surely wrong, as 670b098 is mbed-os-5.2.0-rc1 which I'm assuming shouldn't be visible in the IDE.  And, as in my original report, it hasn't really updated to that version when you compile, but does use that version when you publish to a repo.

screamerbg, 2016-10-07:
@infinnovation Thanks for reporting this. There was a problem on the backend versions of mbed-os. Can you confirm that you're able to compile with the latest release?

Regarding the release candidate - yes, it's supposed to be visible.

infinnovation, 2016-10-11:
mbed-os-blinky-example fails to compile against 2540:670b098

mbed-os-blinky-example after Update - apparently to 2541:a551faa - compiles successfully.

_However_, if you add the line
```C++
#include "network-socket/TCPSocket.h"
```
to main.cpp, it still compiles successfully.  This demonstrates that it is still (or again) using a version from the 5.1 series, since the path network-socket/TCPSocket.h does not exist in the 5.2 tree.

Working offline, if you do "mbed-cli update" in mbed-os, it gets a6f3fd1, tagged mbed-os-5.1.5, latest and mbed_lib_rev127.  This revision does not even appear in the online IDE's list of revisions for mbed-os.

> Regarding the release candidate - yes, it's supposed to be visible.

So, users of the online IDE are used as guinea-pigs for possibly failing release-candidates which users of mbed-cli don't see?

All rather surprising.

screamerbg, 2016-10-11:
> This demonstrates that it is still (or again) using a version from the 5.1 series, since the path network-socket/TCPSocket.h does not exist in the 5.2 tree.

This was caused by the very latest 5.2 not being correctly deployed online. When the Online IDE (and build system) cannot find correctly deployed release/revision, they default to "latest" which in this case is mbed OS 5.1. In the future we should make sure this is more explicit/visible in the IDE.

> Working offline, if you do "mbed-cli update" in mbed-os, it gets a6f3fd1, tagged mbed-os-5.1.5, latest and mbed_lib_rev127. This revision does not even appear in the online IDE's list of revisions for mbed-os.

Thanks for reporting this. The Online IDE was over-filtering non-tagged commits. This is now fixed and available live.

> So, users of the online IDE are used as guinea-pigs for possibly failing release-candidates which users of mbed-cli don't see?

Really sorry that you feel that way, but we don't see the mbed users as "guinea-pigs". All mbed users are able to influence/voice the mbed OS definition and we're happy to provide everyone with visibility on the upcoming mbed OS release. This doesn't mean that we're forcing the beta releases on anyone, but rather provide them if some are looking for these.

infinnovation, 2016-10-12:
It's not that big a deal really.  It's just that I'd expect a simple "Update" to get the latest full release, and if I wanted to to try something nearer the bleeding edge such as a release candidate for the next version then I'd have to choose that explicitly.

I do appreciate the openness of the process, the frequency of releases, the amount of code needed to make it all hang together, and how hard it is to make things easy!

infinnovation, 2016-10-13:
Sorry to harp on about it, but there's another aspect not yet mentioned.  If you import mbed-os-example-blinky with the option "Update this program and libraries to latest revision" ticked, the mbed-os you get is 2246:a6f3fd1.  Which is what I'd personally expect - the latest full release i.e. 5.1.5.  However, the green curly arrow appears, indicating that a further update is available, which gets you 5.2.0-rc2.  This seems a bit inconsistent.

screamerbg, 2016-10-17:
@infinnovation to avoid confusion we've removed the rc releases from the online IDE. Also clicking the Update button gives the latest mbed OS releases. Hope this addresses the consistency concerns you had.

infinnovation, 2016-10-17:
@screamerbg Thanks, it seems a lot more sensible now.

closed 2016-10-17 by infinnovation