ARMmbed/mbed-os #2962: [MAKE EXPORTER] - Signal that srec_cat has to be present if used.

pan-, 2016-10-07:
### Description
Indicate to the user which have exported its project with a makefile exporter that he needs to install [srec_cat](http://srecord.sourceforge.net/) to build its project if their is hex files to merge in it.

- Type: Enhancement 

## Enhancement
On some targets like the `NRF5` family an hex file is merged with the binary as a post build process. While the user doesn't have to install anything special when a project is compiled with mbed cli, it is not the case if the project is exported with the make family exporters. 

In this setup, the makefile use a program named `srec_cat` to merge the hex file(s) and the application, if `srec_cat` is not present on the system, the build end badly like [here](https://github.com/ARMmbed/oob-mbed-os-example-ble/issues/3). 

Instead of an obscure failure it would be better to signal to the user that [`srecord`](http://srecord.sourceforge.net/) have to be installed on its system.

`srec_cat` is only used at [this](https://github.com/ARMmbed/mbed-os/blob/5310c2e2cdfdbee49efa93ef51be5964c58a4549/tools/export/makefile/Makefile.tmpl#L136) place.

0xc0170, 2016-10-07:
cc @theotherjimmy @sarahmarshy

bridadan, 2016-10-07:
Don't the exporters place the following file in the directory? https://github.com/ARMmbed/mbed-os/blob/5310c2e2cdfdbee49efa93ef51be5964c58a4549/tools/export/GettingStarted.html

It contains a link to here: https://developer.mbed.org/handbook/Getting-Started-mbed-Exporters

That page states the following:

## GCC
Makefiles are created and to build you will need make installed and part of your path. Also you will need an arm-gcc compiler and linker.

### Nordic Platforms using SoftDevices
- Windows
    - GCC exports targeting the NRF51822 also require some programs from the Nordic nrf51_SDK. Please download and install it from the nordic website to enable offline compiles with make on windows.
- Linux
    - Need to install xxxx

So maybe that online page could use some updating?

0xc0170, 2016-10-07:
> So maybe that online page could use some updating?

Yes, it should be.

However, we don't check even if compilers exist, so a question becomes, would we do it for this issue, or is it even possible to have it multiplatform? However, as @pan- stated above, makefile could print always a message that ``srecord`` is running and its required to be installed (then it fails and a person can see what's failing) ? It's quite hidden (look at the issue reported there) , so this could be enhancement?

bridadan, 2016-10-07:
> However, we don't check even if compilers exist

Pretty sure we do this as of https://github.com/ARMmbed/mbed-os/pull/2418.

> makefile could print always a message that srecord is running and its required to be installed

Makes sense to me!

0xc0170, 2016-10-07:
>     However, we don't check even if compilers exist

> Pretty sure we do this as of #2418.

I was talking about exported project (makefile in this case). That one does not check, it failed today for mee because my IAR was different than generated by default :-)

bridadan, 2016-10-07:
> I was talking about exported project (makefile in this case). That one does not check, it failed today for mee because my IAR was different than generated by default :-)

Ah thanks, yeah sorry misunderstood :)

pan-, 2016-10-14:
@bridadan Any update on this issue ?

bridadan, 2016-10-14:
@pan- I looked at the makefile template briefly but couldn't immediately pick out where to output this text, especially conditionally for each target. This may be a job for @theotherjimmy when he gets back.

theotherjimmy, 2016-10-18:
@0xc0170 @bridadan 
>> makefile could print always a message that ~~srecord~~ srec_cat is running and its required to be installed

>Makes sense to me!

That's not really supported directly by make, so we would have to worry about the different commands that are needed to find things within the path on the different operating systems, and worry about what they return, and worry about what they will do if you provide an absolute path... I really don't think this is a good idea.

bridadan, 2016-10-18:
@theotherjimmy I think we actually just want to go with the really dumb solution here: Always print the notice about srec :smile: So just a call to `echo`. If we keep it short and sweet, shouldn't be an issue.

theotherjimmy, 2016-10-18:
Actually, you want it to print before compiling right? then it would be a call to `$(info ...)`.

theotherjimmy, 2016-10-18:
@pan- @bridadan See the linked PR.

pan-, 2016-10-18:
Thanks for the PR. That should be enough for the end user and portable across all the OS we support.

ashok-rao, 2016-10-21:
Referring to comments under https://github.com/ARMmbed/oob-mbed-os-example-ble/issues/19 , can we have a confirmation message after the end of srec execution to confirm to the user that the hex file is ready for flashing to target - maybe output the path to the file? the NOTE is a bit mis-leading at the moment.

pan-, 2016-10-21:
@theotherjimmy

theotherjimmy, 2016-10-21:
Yep! re-opening to keep this tracked.

PR #3222 (2016-11-10): [Export - Make] Add message that informs the user which hex to flash
## Description
Resolves #2962 part B in a way that should be noticable

## Reviews
- [x] @0xc0170 
- [x] @pan-
- [x] @ashok-rao 

## Examples
It looks like this (k64f, no hex file merging):
```
$ make -j 
[SNIP]
Compile: port_api.c
Compile: sleep.c
Compile: i2c_api.c
Compile: rtc_api.c
../mbed-os/targets/TARGET_Freescale/TARGET_KSDK2_MCUS/TARGET_MCU_K64F/drivers/fsl_enet.c: In function 'ENET_SetTxBufferDescriptors':
../mbed-os/targets/TARGET_Freescale/TARGET_KSDK2_MCUS/TARGET_MCU_K64F/drivers/fsl_enet.c:493:29: warning: compa [... cut]
closed 2016-11-10 by adbridge