ARMmbed/mbed-os #2965: Exporters - fetching packs from keil.com generates lot of errors

0xc0170, 2016-10-07:
### Description
- Type: Question
- Priority:  Minor

---------------------------------------------------------------
## Bug

**Actual behavior**

Latest master (as of now) generates this:

```
mbed export -i iar -m nrf51_dk
C:\Python27\lib\site-packages\fuzzywuzzy\fuzz.py:35: UserWarning: Using slow pure-python SequenceMatcher. Install python-Levenshtein to remove this warning
  warnings.warn('Using slow pure-python SequenceMatcher. Install python-Levenshtein to remove this warning')
Scan: .
Scan: FEATURE_BLE
Scan: FEATURE_COMMON_PAL
Scan: FEATURE_LWIP
Scan: FEATURE_UVISOR
Scan: FEATURE_LOWPAN_BORDER_ROUTER
Scan: FEATURE_LOWPAN_HOST
Scan: FEATURE_LOWPAN_ROUTER
Scan: FEATURE_NANOSTACK
Scan: FEATURE_NANOSTACK_FULL
Scan: FEATURE_THREAD_BORDER_ROUTER
Scan: FEATURE_THREAD_END_DEVICE
Scan: FEATURE_THREAD_ROUTER
Scan: FEATURE_STORAGE
[ ERROR ] file http://www.keil.com/pack/Keil.ARM_Compiler.pdsc did not download (28, 'Resolving timed out after 2683 milliseconds')
Caching Files 90/160[ ERROR ] file http://www.keil.com/pack/ARM.minar.pdsc did not download (28, 'Resolving timed out after 2683 milliseconds')
[ ERROR ] file http://www.abov.co.kr/data/mds/PACK/ABOV.CM3_DFP.pdsc did not download (28, 'Resolving timed out after 2012 milliseconds')
[ ERROR ] file http://dave.infineon.com/Libraries/CMSIS_PACK/Infineon.TLE986x_DFP.pdsc did not download (28, 'Resolving timed out after 2013 milliseconds')
[ ERROR ] file http://www.micrium.com/cmsis-pack/Micrium.RTOS.pdsc did not download (60, 'SSL certificate problem: self signed certificate in certificate chain')
[ ERROR ] file https://realtimelogic.com/downloads/arm-mdk-packs/RealTimeLogic.SMQ.pdsc did not download (60, 'SSL certificate problem: unable to get local issuer certificate')
[ ERROR ] file http://developer.nordicsemi.com/nRF51_SDK/pieces/nRF_Libraries/NordicSemiconductor.nRF_Libraries.pdsc did not download (28, 'Resolving timed out after 2013 milliseconds')
[ ERROR ] file http://developer.nordicsemi.com/nRF51_SDK/pieces/nRF_Drivers_External/NordicSemiconductor.nRF_Drivers_External.pdsc did not download (28, 'Resolving timed out after 2012 milliseconds')
[ ERROR ] file http://developer.nordicsemi.com/nRF51_SDK/pieces/nRF_BLE/NordicSemiconductor.nRF_BLE.pdsc did not download (28, 'Resolving timed out after 2012 milliseconds')
[ ERROR ] file http://developer.nordicsemi.com/nRF51_SDK/pieces/nRF_ANT/NordicSemiconductor.nRF_ANT.pdsc did not download (28, 'Resolving timed out after 2012 milliseconds')
[ ERROR ] file http://www.mindmotion.com.cn/Download/MDK_KEIL/MindMotion.MM32x103_DFP.pdsc did not download (28, 'Connection timed out after 2013 milliseconds')
[ ERROR ] file http://www.mindmotion.com.cn/Download/MDK_KEIL/MindMotion.MM32x031_DFP.pdsc did not download (28, 'Resolving timed out after 2012 milliseconds')
[ ERROR ] file http://download.labs.mediatek.com/MediaTek.MTx.pdsc did not download (28, 'Resolving timed out after 2012 milliseconds')
[ ERROR ] file https://realtimelogic.com/downloads/arm-mdk-packs/RealTimeLogic.SharkSSL-Lite.pdsc did not download (28, 'Resolving timed out after 2012 milliseconds')
[ ERROR ] file http://software-dl.ti.com/msp430/msp430_public_sw/mcu/msp430/mspkeil/latest/exports/TexasInstruments.MSP432P4xx_DFP.pdsc did not download (28, 'Resolving timed out after 2512 milliseconds')
[ ERROR ] file http://liveupdate.sonix.com.tw/sonix/develop_tool/MCU/DFP/SONiX.SN32F7_DFP.pdsc did not download (28, 'Resolving timed out after 2013 milliseconds')
[ ERROR ] file http://www.abov.co.kr/data/mds/PACK/ABOV.CM0_DFP.pdsc did not download (28, 'Resolving timed out after 2012 milliseconds')
[ ERROR ] file http://liveupdate.sonix.com.tw/sonix/develop_tool/MCU/DFP/SONiX.SN32F1_DFP.pdsc did not download (28, 'Resolving timed out after 2012 milliseconds')
[ ERROR ] file http://s3.asia.ambiqmicro.com/pack/AmbiqMicro.Apollo_DFP.pdsc did not download (28, 'Resolving timed out after 2013 milliseconds')
[ ERROR ] file http://www.nuvoton.com/hq/enu/Documents/KEILSoftwarePack/Nuvoton.NuMicro_DFP.pdsc did not download (28, 'Operation too slow. Less than 51200 bytes/sec transferred the last 2 seconds')
Caching Files 161/160
[ ERROR ] file http://www.keil.com/pack/ARM.minar.pdsc
'NoneType' object has no attribute 'url'
[ ERROR ] file http://www.keil.com/pack/Keil.ARM_Compiler.pdsc
'NoneType' object has no attribute 'url'
[ ERROR ] file http://www.abov.co.kr/data/mds/PACK/ABOV.CM3_DFP.pdsc
'NoneType' object has no attribute 'url'
[ ERROR ] file http://dave.infineon.com/Libraries/CMSIS_PACK/Infineon.TLE986x_DFP.pdsc
'NoneType' object has no attribute 'url'
[ ERROR ] file http://developer.nordicsemi.com/nRF51_SDK/pieces/nRF_Libraries/NordicSemiconductor.nRF_Libraries.pdsc
'NoneType' object has no attribute 'url'
[ ERROR ] file http://developer.nordicsemi.com/nRF51_SDK/pieces/nRF_Drivers_External/NordicSemiconductor.nRF_Drivers_External.pdsc
'NoneType' object has no attribute 'url'
[ ERROR ] file http://developer.nordicsemi.com/nRF51_SDK/pieces/nRF_BLE/NordicSemiconductor.nRF_BLE.pdsc
'NoneType' object has no attribute 'url'
[ ERROR ] file http://developer.nordicsemi.com/nRF51_SDK/pieces/nRF_ANT/NordicSemiconductor.nRF_ANT.pdsc
'NoneType' object has no attribute 'url'
[ ERROR ] file http://www.mindmotion.com.cn/Download/MDK_KEIL/MindMotion.MM32x103_DFP.pdsc
'NoneType' object has no attribute 'url'
[ ERROR ] file http://www.mindmotion.com.cn/Download/MDK_KEIL/MindMotion.MM32x031_DFP.pdsc
'NoneType' object has no attribute 'url'
[ ERROR ] file http://www.micrium.com/cmsis-pack/Micrium.RTOS.pdsc
'NoneType' object has no attribute 'url'
[ ERROR ] file http://download.labs.mediatek.com/MediaTek.MTx.pdsc
'NoneType' object has no attribute 'url'
[ ERROR ] file http://software-dl.ti.com/msp430/msp430_public_sw/mcu/msp430/mspkeil/latest/exports/TexasInstruments.MSP432P4xx_DFP.pdsc
'NoneType' object has no attribute 'url'
[ ERROR ] file http://www.wolfSSL.com/files/ide/wolfSSL.CyaSSL.pdsc
'NoneType' object has no attribute 'url'
[ ERROR ] file https://realtimelogic.com/downloads/arm-mdk-packs/RealTimeLogic.SMQ.pdsc
'NoneType' object has no attribute 'url'
[ ERROR ] file https://realtimelogic.com/downloads/arm-mdk-packs/RealTimeLogic.SharkSSL-Lite.pdsc
'NoneType' object has no attribute 'url'
[ ERROR ] file http://www.wolfSSL.com/files/ide/wolfSSL.wolfSSL.pdsc
'NoneType' object has no attribute 'url'
[ ERROR ] file http://liveupdate.sonix.com.tw/sonix/develop_tool/MCU/DFP/SONiX.SN32F7_DFP.pdsc
'NoneType' object has no attribute 'url'
[ ERROR ] file http://liveupdate.sonix.com.tw/sonix/develop_tool/MCU/DFP/SONiX.SN32F1_DFP.pdsc
'NoneType' object has no attribute 'url'
[ ERROR ] file http://www.abov.co.kr/data/mds/PACK/ABOV.CM0_DFP.pdsc
'NoneType' object has no attribute 'url'
[ ERROR ] file http://s3.asia.ambiqmicro.com/pack/AmbiqMicro.Apollo_DFP.pdsc
'NoneType' object has no attribute 'url'
Generating Index 160/160
Scanning for Aliases 160/160
```

Is this expected?

**Steps to reproduce**

I just run ``mbed export -i iar -m nrf51_dk`` and shown above output. 

cc @sarahmarshy @theotherjimmy

bridadan, 2016-10-07:
When I did this on my machine I also saw these errors. I think the good news is it shouldn't affect exporting. Maybe we could suppress the errors unless the `-v` flag is provided. Does that sound like a good option @0xc0170 ?

0xc0170, 2016-10-07:
Yes, I forgot to state, exporting works, just that there's lot of errors that I believe could be fixed or suppressed (I don't know how important are those and why they are displayed).

theotherjimmy, 2016-10-20:
Fetching is no longer done. Closing.

closed 2016-10-20 by theotherjimmy