ARMmbed/mbed-os #2968: User defines in Makefile

nuket, 2016-10-10:
### Description
- Type: Enhancement
- Related issue: `#2927`
- Priority: Minor

----------------------------------------------------------------
## Enhancement

Might be useful if there were C_FLAGS_EXTRA, CXX_FLAGS_EXTRA and ASM_FLAGS_EXTRA parameters that could be tacked on to the end of the various *_FLAGS in the generated Makefile, so that one could then run `make C_FLAGS_EXTRA=...` etc. to pass along some parameters.

i.e. `C_FLAGS += $(C_FLAGS_EXTRA)` and so on.

0xc0170, 2016-10-10:
cc @theotherjimmy

geky, 2016-10-28:
Does `make CFLAGS+=...` work?

theotherjimmy, 2016-11-02:
Nope.

theotherjimmy, 2016-11-02:
HAHA! I got it: `C_FLAGS=<whatever> make`

0xc0170, 2016-11-03:
@theotherjimmy What do you propose? To use this without code changes or do code changes?

theotherjimmy, 2016-11-03:
I propose that we use the this code as is.

theotherjimmy, 2016-12-20:
@nuket I think the proposed `CFLAGS=<extra-stuff> make` solves your problem. Alternatively, you could export with `--profile <your-profile>` to overwrite the cflags. If this does not solve your problem please feel free to re-open this issue.

closed 2016-12-20 by theotherjimmy