ARMmbed/mbed-os #2971: [Nucleo L073RZ] Flash load region size is not correct

toyowata, 2016-10-10:
### Description
- Type: Bug
- Priority: Major

---------------------------------------------------------------
## Bug

The problem reported at the Questions in developer site:
https://developer.mbed.org/forum/bugs-suggestions/topic/26719/

>When I try to compile my program for the platform Nucleo L073RZ with the online IDE, I get this error:
>
>Error: Load region LR_IROM1 size (69108 bytes) exceeds limit (65536 bytes). Region contains 105 bytes of padding and 0 bytes of veneers (total 105 bytes of linker generated content).
>
>However, when I export to the gcc toolchain everything goes fine.
>
>Nucleo L053R8 has a 64KB Flash, whilst Nucleo L073RZ has a 192KB Flash.
>
>May it be an error defining the flash size in the linker file used by the online compiler?

**Target**
NUCLEO-L073RZ

**Toolchain:**
ARM

**Toolchain version:**
Online compiler (armcc 5.06)

**meed-os sha:**
mbed library v127

**Expected behavior**
No link error

**Actual behavior**
Link error L6220E

**Steps to reproduce**
Please use the test code below:
```
#include "mbed.h"

DigitalOut myled(LED1);
Serial pc(USBTX, USBRX);

#pragma no_inline

#if defined(TARGET_LPC1114) || defined(TARGET_SSCI824)
int foo1(int x) __attribute__((section(".ARM.__at_0x00000FF8")));
int foo2(int x) __attribute__((section(".ARM.__at_0x00004004")));
int foo3(int x) __attribute__((section(".ARM.__at_0x00007800")));
#elif defined(TARGET_HRM1017) || defined(TARGET_TY51822R3)
int foo1(int x) __attribute__((section(".ARM.__at_0x0001C200")));
int foo2(int x) __attribute__((section(".ARM.__at_0x00020000")));
int foo3(int x) __attribute__((section(".ARM.__at_0x00023F00")));
#elif defined(TARGET_LPC4337)
int foo1(int x) __attribute__((section(".ARM.__at_0x1A010000")));
int foo2(int x) __attribute__((section(".ARM.__at_0x1A020000")));
int foo3(int x) __attribute__((section(".ARM.__at_0x1A012000")));
#elif defined(TARGET_LPC11U68)
int foo1(int x) __attribute__((section(".ARM.__at_0x00018000")));
int foo2(int x) __attribute__((section(".ARM.__at_0x00020000")));
int foo3(int x) __attribute__((section(".ARM.__at_0x00011000")));
#elif defined(TARGET_LPC1549)
int foo1(int x) __attribute__((section(".ARM.__at_0x00020000")));
int foo2(int x) __attribute__((section(".ARM.__at_0x00030000")));
int foo3(int x) __attribute__((section(".ARM.__at_0x0003B000")));
#elif defined(TARGET_RZ_A1H)
int foo1(int x) __attribute__((section(".ARM.__at_0x18004000")));
int foo2(int x) __attribute__((section(".ARM.__at_0x18005000")));
int foo3(int x) __attribute__((section(".ARM.__at_0x18006000")));
#elif defined(TARGET_LPC11U35_501)
int foo1(int x) __attribute__((section(".ARM.__at_0x00001000")));
int foo2(int x) __attribute__((section(".ARM.__at_0x00008000")));
int foo3(int x) __attribute__((section(".ARM.__at_0x0000F800")));
#elif defined(TARGET_NUCLEO_L073RZ)
// 0x08000000 - 0x08030000
int foo1(int x) __attribute__((section(".ARM.__at_0x08001000")));
int foo2(int x) __attribute__((section(".ARM.__at_0x08010000")));
int foo3(int x) __attribute__((section(".ARM.__at_0x0802F000")));
#else
int foo1(int x);
int foo2(int x);
int foo3(int x);
#endif

int main() {
    pc.printf("\n*** test start ***\n");
    pc.printf("Test : foo1 = %d\n", foo1(123));
    pc.printf("Test : foo2 = %d\n", foo2(123));
    pc.printf("Test : foo3 = %d\n", foo3(123));
    pc.printf("*** test end ***\n");
    while(1) {
        myled = 1;
        wait(0.2);
        myled = 0;
        wait(0.2);
    }
}

int foo1(int x)
{
    return (x+1);
}

int foo2(int x)
{
    return (x+2);
}

int foo3(int x)
{
    return (x+3);
}

```

I guess the load region size is not correct in the scatter file.
https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_STM/TARGET_STM32L0/TARGET_NUCLEO_L073RZ/device/TOOLCHAIN_ARM_STD/stm32l073xz.sct

```
LR_IROM1 0x08000000 0x10000  {    ; load region size_region
```
This should be:
```
LR_IROM1 0x08000000 0x30000  {    ; load region size_region
```

jeromecoutant, 2016-10-10:
Hi
This is already corrected in https://github.com/ARMmbed/mbed-os/pull/2967
Thx

toyowata, 2016-10-11:
Closed by merging #2967

closed 2016-10-11 by toyowata