ARMmbed/mbed-os #2978: [NUC472] tests-mbedtls-selftest fails with uARM

ccli8, 2016-10-11:
### Description
- Type: Bug
- Priority: Minor

---------------------------------------------------------------
## Bug

**Target**
NUMAKER_PFM_NUC472

**Toolchain:**
uARM

**Toolchain version:**
MDK-ARM 5.17

**mbed-cli version:** 
0.9.5

**meed-os sha:**
f5fb485

The sample `tests-mbedtls-selftest` fails with uARM toolchain. It is passed with ARM, GCC_ARM, and IAR toolchains. If I enlarge stack size from 128 to 256 (`mbed-os/nu-porting/mbed/rtos/rtx/TARGET_CORTEX_M/cmsis_os.h`), it is fixed.
```
#if defined(TOOLCHAIN_GCC) || defined(TOOLCHAIN_ARM_STD) || defined(TOOLCHAIN_IAR)
#    define WORDS_STACK_SIZE   512
#elif defined(TOOLCHAIN_ARM_MICRO)
#    define WORDS_STACK_SIZE   128
#endif
```

0xc0170, 2016-10-11:
cc @andresag01 @yanesca

andresag01, 2016-10-11:
@sbutcher-arm

sg-, 2017-01-19:
mbed OS 5 doesnt support uARM. Can we just delete all the macros and set the stack size to 512?

sbutcher-arm, 2017-02-13:
Hi @ccli8,

My colleague @andresag01 is looking into this, and we hope to have a resolution shortly.

andresag01, 2017-02-14:
@ccli8, @sbutcher-arm, @sg-: I had a look at this issue and found that the test fails because of limited stack memory as previously reported. I doubled the stack size as suggested (i.e. from 128 to 256) and the test executes correctly, but RTX still reports a stack overflow error. I doubled the stack size again (i.e. from 256 to 512) and the test works as expected.

The simple solution is to increase the stack size to 512. However, after speaking with @0xc0170, it is difficult to justify this change for the whole of mbed OS to fix a single test. An alternative solution is to modify RTX so that we can configure the stack size through a `mbed_app.json` for that specific test. I started working on a patch to enable this, but it is not possible to compile a single test with a specific `mbed_app.json` and still compile the remaining tests with a different `mbed_app.json`. I have raised a separate issue to report this problem: https://github.com/ARMmbed/mbed-os/issues/3770.

**NOTE:** It is possible to pass the `--app-config` flag to `mbed test`, but this will compile all tests with a `mbed_app.json`. This will work in my local machine, but the test will still fail in the CI.

ccli8, 2017-06-22:
I check the issue has been fixed on mbed-os-5.5.0.

closed 2017-06-22 by ccli8