ARMmbed/mbed-os #2980: Nordic nRF51822 + GNU ARM Embedded Error

iotvietmember, 2016-10-11:
[image]

Hello. I am using Nordic nRF51822 board and I created a simple project (mbed_blinky). Atfer that, I exported it to GCC (ARM Embedded).
I created a project on Eclipse (with GNU ARM Eclipse plugin).
I build project by make and I met an error: " make VERBOSE=1 all 
c:/program files (x86)/gnu arm eclipse/build tools/2.6-201507152002/bin/sh: syntax error: unexpected "("
Makefile:21: recipe for target 'all' failed
make: *** [all] Error 2"
Please helps me.
Many thanks

0xc0170, 2016-10-11:
cc @pan- @sarahmarshy @theotherjimmy

pan-, 2016-10-12:
@iotvietmember Did you try to run make on the command line ? 
Gnu ARM eclipse comes with its own version of `sh` (a busybox one), this might be the cause of your troubles.

iotvietmember, 2016-10-12:
Yes. I tried to run make on the command line but it is same error

[image]

iotvietmember, 2016-10-12:
[image]

pan-, 2016-10-12:
@iotvietmember Your make still invoke sh from gnu arm tools (`/bin/sh`). It would have been nice to try without it (just move it temporarily somewhere else). 
Another thing, could you print the content of `OBJDIR` ?

iotvietmember, 2016-10-12:
@pan- 
Yes. You are right. 
The problem is at "sh.exe ".
Everything is ok.
Thank you very much!

closed 2016-10-12 by iotvietmember

pan-, 2016-10-12:
@iotvietmember Well, the issue is still there. GNU ARM eclipse is not supported oob by mbed but you can open a ticket requesting this enhancement.

iotvietmember, 2016-10-12:
@pan-  Thanks. Can I send a request to GNU ARM Eclipse team?

sarahmarshy, 2016-10-12:
It's not? You should be able to export to eclipse_armc5. Then, you,can import your project directory as an existing project in eclipse.

pan-, 2016-10-12:
@iotvietmember I think the request should be there if you want to make one.

@sarahmarshy GNU ARM eclipse != DS5. But to be honest I didn't try the exporter.

theotherjimmy, 2016-10-12:
@pan- eclipse_armc5 != DS-5. You only need eclipse and armc5, not all of DS-5

iotvietmember, 2016-10-12:
Yes. My error is sh.exe of GNU Arm eclipse Build Tools. 
Thank you very much