ARMmbed/mbed-os #2983: mbedtls self test not calling mbedtls_sha1_self_test ?

adustm, 2016-10-11:
### Description
- Type: Question

Hello, Is there any particular reason that the TESTS/mbedtls/selftest/main.cpp is not calling the SHA 1 self test ?
Cheers
Armelle

0xc0170, 2016-10-11:
cc @andresag01 @sbutcher-arm

andresag01, 2016-10-11:
@adustm: We have only added a very small subset of tests to the self test. Our main motivation was to add entropy. At some point, it would be ideal to add more tests. However, I would still like to point out that even if the code for SHA1 is added, the test will be excluded because in [feature/mbedtls MBEDTLS_SHA1_C is disabled by default](https://github.com/ARMmbed/mbed-os/blob/master/features/mbedtls/inc/mbedtls/config.h#L2220)

closed 2017-01-19 by sg-