ARMmbed/mbed-os #2989: spi_format() bug on KL34Z/KL27Z

DrSchottky, 2016-10-11:
### Description
- Type: Bug 
- Priority:  Major

---------------------------------------------------------------
## Bug
`void spi_format(spi_t *obj, int bits, int mode, int slave)` in spi_api.c always fails due to an incorrect check on `bits` arg.

`if ((bits != 8) || (bits != 16)) 
{
    error("Only 8bits and 16bits SPI supported");
    return;
}`

evaluates to true whatever the value of `bits` is.

**Target**
 KL34Z KL27Z

andrewwade, 2017-02-19:
I found the same error.

0xc0170, 2017-02-20:
> I found the same error. Should be && instead of | |

Thanks, lets fix it now

0xc0170, 2017-02-20:
A fix was sent (please see it above)

DrSchottky, 2017-02-20:
Similar flaw found in TARGET_K20XX/spi_api.c, line 60
`MBED_ASSERT((bits > 4) || (bits < 16));`

PR #3806 (2017-02-21): NXP KL43Z/KL27Z: fix spi format bits check
It was always true for valid values (if its not 8 neither 16 bits, fail).

Fixes #2989. Logical error. Tested - compile tests for these 2 platforms.
commit bf778576 (2017-02-21): NXP KL43Z/KL27Z: fix spi format bits check

Fixes #2989. It was always true for valid values (if its not 8 neither 16 bits, fail).
commit 6e4d261d (2017-02-24): NXP KL43Z/KL27Z: fix spi format bits check

Fixes #2989. It was always true for valid values (if its not 8 neither 16 bits, fail).
closed 2017-02-21 by 0xc0170